360Controller/360Controller #789: Controller is recognized, but in a few seconds it lights all four lights and doesn't respond anymore.

buykkothen, 2018-11-01:
### Type of Controller
Wired 360

### OS Version
10.13.6

### Driver Version
Latest

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
N/A

### Response of Controller
The guide button light "player 1" for some seconds, then it lights all players lights and stop working.

### Response in Preference Pane
The controller is recognized but don't show any button response in the extension.

### Detailed Explaination of Issue
The guide button light "player 1" for some seconds, then it lights all players lights and stop working. It all started 3/4 days ago, it never had this before. I have to unplug and plug it again for it to work for some minute or two before it happens again.

FranticRain, 2018-11-01:
Controller information is only invalid for things like typos in the preference pane or something that doesn't involve the controller at all. Please retrieve your controller info and let me know when you have updated your post.

buykkothen, 2018-11-05:
Product ID: 0x028e
Vendor ID: 0x045e  (Microsoft Corporation)

FranticRain, 2018-11-05:
Please run this command in your terminal, enter your password when prompted, and tell me what the output is:
```
sudo kextutil /Library/Extensions/360Controller.kext
```

buykkothen, 2018-11-06:
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

FranticRain, 2018-11-07:
Can you confirm with a windows machine or Xbox that this controller is still working properly? It looks like everything should be working as intended from the driver side.

buykkothen, 2018-11-07:
Yep, working on Windows Machine, the difference is that in windows it doesn't stop working suddenly =/

buykkothen, 2018-11-09:
Hello.

I've noticed a small difference while using the 0.16.8 software, it seems to disconnect less frequently and works better. But still disconnecting sometimes.