360Controller/360Controller #790: Controller unresponsive / undetected in driver

games1theuniverse, 2018-11-04:
### Type of Controller
N/A (afterglow wired)

### OS Version
Latest version of Mac OS Mojave on a mid 2015 macbook pro (contact me for specs)

### Driver Version
0.16.10

### Connection Method
Wired, And yes, i'm using a data wire

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Afterglow Wired Controller for Xbox One
<br/>Product ID: 0x0139
<br/>Vendor ID: 0x0e6f

### Response of Controller
it just sits there and glows like it does when connected to anything (yes even a wall)

### Response in Preference Pane
shows the name (afterglow wired controller)

### Detailed Explaination of Issue
So i am able to connect it via usb just fine and it glows but the driver doesnt detect it or work in any games, i have also tried it in the browser, Buttons and joysticks all unresponsive,

FranticRain, 2018-11-04:
Afterglow controllers fall under the PDP umbrella and are unsupported by the driver. As of 10.11, macOS changed their USB code in the operating system that shows itself as various different issues. Such as the controllers not being recognized, or when they are loaded into the driver, can't get any data out of them.

As your controller is being recognized by the driver, I'm just going to tag it with the issue and leave this open. It will probably never be resolved without outside assistance. If you would like to get this solved, you would need to open a developer support request and send them your controller so that they could figure out how they broke their code in the shim.

It might also be fixed in the controller rewrite, where we update the driver to the modern framework. But at this time, that is not planned.

FranticRain, 2019-04-28:
Could you give the 1.0.0-alpha.1 version of the driver a shot and see if it solves this problem?

FranticRain, 2019-10-28:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-10-28 by FranticRain