360Controller/360Controller #791: Wired Controller not recognized by driver

TobiasWaslowski, 2018-11-04:
### Type of Controller
Replace this text with: Xbox 360

### OS Version
Mac OS Mojave 10.14

### Driver Version
0.16.10

### Connection Method
Wired

### Device Name and Info

>Device Name: Gamepad for Xbox 360
<br/>Product ID: 0x0301
<br/>Vendor ID: 0x0e6f

### Response of Controller
When connected to the PC, the center button lights up for about one second. After that, nothing happens.

### Response in Preference Pane
The preference pane shows "No Devices found". Pressing the buttons of the controller does nothing.

### Detailed Explaination of Issue
Installed the Tattiebogle driver, found that it wasn't working with my controller and offered little troubleshooting help, so I installed this one (and uninstalled Tattiebogle + rebooted). So far, I have only tested it on the 0.16.10. After checking some of the other issue threads, I tried running 
sudo kextutil /Library/Extensions/360Controller.kext
with the following output:

Notice: /Library/Extensions/360Controller.kext has debug properties set.
Kext rejected due to system policy: <OSKext 0x7fd3ca494f30 [0x7fff98da58e0]> { URL = "file:///Library/StagedExtensions/Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Kext rejected due to system policy: <OSKext 0x7fd3ca494f30 [0x7fff98da58e0]> { URL = "file:///Library/StagedExtensions/Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Diagnostics for /Library/Extensions/360Controller.kext:
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

Thank you in advance!

FranticRain, 2018-11-04:
Please ensure that the driver has been allowed in System Preferences -> Security and Privacy -> General. This approval UI is only present in the Security & Privacy preferences pane for 30 minutes after the alert. Until you approve the kext, future load attempts will cause the approval UI to reappear but will not trigger another user alert.

[image]

TobiasWaslowski, 2018-11-05:
Yep, that did the job. Thanks a lot!

closed 2018-11-05 by FranticRain