360Controller/360Controller #793: Xbox One afterglow controller, no devices found

GreenGiffin, 2018-11-10:
### Type of Controller
Afterglow Xbox One

### OS Version
Mojave 10.14.1

### Driver Version
360Controller 0.16.10

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
Afterglow Wired Controller for Xbox One:
Product ID:	0x02b8
Vendor ID:	0x0e6f

### Response of Controller
The internal lights of the controller light up and stay lit. The main Xbox button in the top center doesn't light up. 

### Response in Preference Pane
Not exactly sure. It just says "no device found".

### Detailed Explanation of Issue
I have followed all the steps in previous issues threads and still can't get my MacBook Pro to recognize the wired controller. It does show up in the System Report under "USB", but the driver isn't recognizing it
[image]
[image]
[image]
[image]

.

GreenGiffin, 2018-11-13:
> You would have to use a version < 10.11.

Is there an Xbox controller you would recommend for me to be able to do but I’m trying to do?

FranticRain, 2018-11-13:
I have no experience with third party controllers and I always recommend the official ones. Especially the most recent one, as it is natively supported via macOS through its Bluetooth connectivity.

GreenGiffin, 2018-11-13:
What’s the name of the one you are recommending or the controller you have the least issues with?

FranticRain, 2018-11-15:
This driver does not make the controller act like a mouse. It doesn't touch the mouse or display code in any way. My best suggestion to you is to check your display layout in your System Preferences and make sure that your displays are located where you think they should be. Anything beyond that is outside the scope of support for this driver.