360Controller/360Controller #796: Razer Wolverine Ultimate light up and rumbles but is otherwise irresponsive

alexkerscher, 2018-11-17:
### Type of Controller
Replace this text with: Razer Wolverine Ultimate

### OS Version
10.14

### Driver Version
0.16.10

### Connection Method
Wired Controller with a Proprietary Razer USB cord connected on a USB-C port through a USB to USB-C Adapter.

### Device Name and Info
>Device Name: Razer Wolverine Ultimate
<br/>Product ID: 0x0a14
<br/>Vendor ID: 0x1532

### Response of Controller
LED lights up and it rumbles at plug in. Lower bar buttons which are for internal usage of the controller seem to be responsive, but all other buttons are irresponsive. 

### Response in Preference Pane
I see this one entry that says "1: Xbox 360 Wired Controller (Wired)". Other than that nothing else happens when I mess around with the controller or anything.

### Detailed Explanation of Issue
Hi team. First and foremost thanks for making this awesome driver.

So, what's going on is that I installed the driver, restarted the computer, plugged in the controller which rumbled and lit up like an x-mas tree and when I went in to check the configurations on the preference pane it didn't had any response when I messed around with the controller pressing it's buttons or moving the analog stick.

When I look into the System Preferences I don't see a device called "Controller" plugged in on the USB section, just one that says Razer Wolverine Ultimate, and another one that says iBridge. 

One thing worth noting is that the controller is plugged into the Macbook's USB-C port using a USB to USB-C adapter at the end of its proprietary Razor cord and I don't know how this could affect the performance of the driver or anything - I have looked into the readme file and it says something about a data cable power cable situation and from what I can tell the adapter has worked as intended in the past to connect pen drives into this Macbook, and a wired keyboard on a raspberry pi.
.
Again, thanks for the great work and for the amazing support that you guys extend to the community. 

Cheers,

Alex.

FranticRain, 2018-11-18:
Duplicate of #722. As mentioned there, a packed sniff will need to be completed before this device can be supported.

FranticRain, 2019-05-26:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-05-26 by FranticRain