360Controller/360Controller #797: No power to the controller

petotePal, 2018-11-17:
### Type of Controller
PDP X Box One Controller, Crimson Red

### OS Version
10.12.6 (16G29)

### Driver Version
latest, 360ControllerInstall_0.16.10
### Connection Method
Wired

### Device Name and Info
PDP Wired Controller for Xbox One - Crimson Red:

  Product ID:	0x02a2
  Vendor ID:	0x0e6f
  Version:	1.0f
  Serial Number:	0000C017BEB5CA11
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Performance Designed Products
  Location ID:	0x14100000 / 6
  Current Available (mA):	500
  Extra Operating Current (mA):	0

### Response of Controller
Nothing at all. I believe when i first plugged the controller in via USB a red LED within the controller lit up briefly, but no further response from the controller.

### Response in Preference Pane
No device recognised.

### Detailed Explaination of Issue
I purchased the controller today after researching controllers for Mac online, from a reputable major retailer in Australia. It's a PDP Wired Controller 'For X-Box One and Windows' so I believe It's a legit and licensed product. I plugged it into my Macbook Air using the included USB cord, and using system profiler saw that It was identified. I then downloaded and installed the latest version of your software, restarted the computer. Unfortunately this is where we reach the impasse, as I then plugged the controller In again after restart, but nothing happens. I press the 'big-X' button, plus every button and nothing happens, no lights on the controller, no messages from Mac OS. Tried both USB inputs, It's still recognised In system profiler but that's It I'm afraid. Have tried restarting several times and re-installing the software. Have looked at the extensions tab in System preferences, no extension s added.

Many thanks for your help,

Alex

PeyotePal, 2018-11-17:
So reading through some other requests I'm starting to assume PDP controllers aren't supported? Shame, at least I bought It on special sigh

FranticRain, 2018-11-18:
Here is a build with your controller added. It may or may not work.

[360ControllerInstall.dmg.zip](https://github.com/360Controller/360Controller/files/2592600/360ControllerInstall.dmg.zip)

closed 2018-11-18 by FranticRain

PeyotePal, 2018-11-18:
Hi Drew,

After a couple of restarts It does indeed work, thankyou so much, amazing
dedication and generosity,

Alex,

www.soundcloud.com/seagram-murals

PeyotePal, 2018-11-19:
Just thought I'd mention I also found this little app that allows Mac users
to map keys to their X-Box controller (and works with your program)
https://yukkurigames.com/enjoyable/

Having destroyed my Shift keys playing Zen Pinball 2 It's allowed me to
start playing again!

Thanks again,

Alex