360Controller/360Controller #798: Bluetooth Support Please?

AdautoP, 2018-11-17:
Please add bluetooth connection to your driver. It doesn't recognizes the controller if you are connected by bluetooth. I know that Mac supports Xbox One S bluetooth connection natively, but it doesn't map the keybindings like your driver does.

FranticRain, 2018-11-18:
This is a duplicate of #747. I would please ask that you read and consider what I have posted in that thread. What you are asking is infeasible. Consider this. An operating system connects to a Bluetooth device. This Bluetooth device advertises itself as a HID compliant device and matches itself to the generic driver. Done. What you are asking is for the operating system to pair to a Bluetooth device, have a driver intercept a standard HID device, process the data being sent into an Xbox 360 controller format, and then return that data back to the OS saying that it is, instead, a USB connection. Without this process, this driver will not work for the absolutely insane methods Developers have built to use the Xbox 360 controller in their games. Developers literally do not support the standard controller interface in the operating system and that's insane.

closed 2018-11-18 by FranticRain

AdautoP, 2018-11-18:
Oh, makes sense! Im so sorry then. And thanks for the response. Looks like
now I have to ask the games developers to add the support, once the driver
is already there.

>