360Controller/360Controller #800: Controller not recognized in driver

buykkothen, 2018-11-19:
### Type of Controller
XBOX 360

### OS Version
10.13.6
### Driver Version
0.16.8

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: USB Joystick
<br/>Product ID: 0x0006
<br/>Vendor ID: 0x0079

### Response of Controller
Any response

### Response in Preference Pane
"Device not found"
### Detailed Explaination of Issue
Controller is recognized by system report USB tab, but not found by the 360controller driver

FranticRain, 2018-11-20:
Is this device supported by official Xbox hardware such as the Xbox One or Xbox 360 consoles?

buykkothen, 2018-11-22:
i don't know, i have never tried because i don't have an xbox, but i think it's for PCs only

nettemple, 2018-11-25:
Having same issue - I've tried different cables as well. Xbox controller shows in System report, but is not recognized in controller driver or in system privacy and Security areas... I am running 10.14(915) on iMac

markhicken, 2018-11-26:
I had to disconnect the controller from Bluetooth and tell the Mac to ignore the device. Then after a reboot, it recognized a wired connection and works for me.

FranticRain, 2018-11-26:
@nettemple Make a new issue. Unless you have an identical device, your issue is unrelated.

@markhicken This is unrelated to the issues at hand.

I need to make a build of the driver that includes this device to see if it works, but I haven't had a chance yet. Please stop piling on an unrelated issue.

FranticRain, 2018-12-06:
@buykkothen Here's a build of the driver with this controller added. Let me know if your controller works or not.
[Install360Controller.pkg.zip](https://github.com/360Controller/360Controller/files/2651553/Install360Controller.pkg.zip)

gerab, 2018-12-23:
@buykkothen I have same device id/vendor id and this build didn't help

FranticRain, 2018-12-26:
Then this is confirmed a counterfeit device and is not supported by this driver. Closing accordingly.

closed 2018-12-26 by FranticRain