360Controller/360Controller #803: Controller Unresponsive

Motikop, 2018-11-25:
### Type of Controller
Xbox One Elite

### OS Version
Mojave 10.14.1

### Driver Version
Version 0.16.10

### Connection Method
Wired (MicroUSB - USB - USBC)

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Controller
<br/>Product ID: 0x02e3
<br/>Vendor ID:0x045e

### Response of Controller
Vibrates when connected to mac,

### Response in Preference Pane
Does not appear in the preference pane (No Devices Found)

### Detailed Explaination of Issue
Worked before perfectly on my 2015 macbook pro, but my new mac only has usbc, so im using the same cable, but pluging the usb into an apple iBridge usb - usbc then connecting it to my computer. It appears under system information as controller, but not in the driver.

FranticRain, 2018-12-03:
The driver is known have issues with USB-C dongles, but I want to make sure that that is your problem. Please run:
```
sudo kextutil /Library/Extensions/360Controller.kext
```
in your Terminal and enter your password as prompted. This will try and manually load the driver and tell me if something is wrong there.

Motikop, 2018-12-23:
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Kext rejected due to system policy: <OSKext 0x7ffe566247a0 [0x7fffa5c948f0]> { URL = "file:///Library/StagedExtensions/Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Kext rejected due to system policy: <OSKext 0x7ffe566247a0 [0x7fffa5c948f0]> { URL = "file:///Library/StagedExtensions/Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Diagnostics for /Library/Extensions/360Controller.kext:
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

FranticRain, 2019-01-02:
Please ensure that the driver has been allowed in System Preferences -> Security and Privacy -> General. This approval UI is only present in the Security & Privacy preferences pane for 30 minutes after the alert. Until you approve the kext, future load attempts will cause the approval UI to reappear but will not trigger another user alert.

[image]

FranticRain, 2019-04-06:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-04-06 by FranticRain