360Controller/360Controller #807: Installed Latest version - and after reboot isn't installed ... ???

L0VEFIRE, 2018-12-03:
### Type of Controller
Replace this text with: Original Xbox, Xbox 360, Xbox One, All or N/A

### OS Version
Replace this text with the version of macOS that you are running

### Driver Version
Replate this text with the version or versions of the driver that you are using or have tested

### Connection Method
Replace this text with: Wired, Wireless Adapter, Bluetooth, N/A or more specific method if applicable

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID:
<br/>Vendor ID:

### Response of Controller
Replace this text with information on how the physical controller reacts. Does the guide button light up? Does it vibrate? Etc.

### Response in Preference Pane
Replace this text with information about the controller's reaction in the preference pane. Does the controller appear in the preference pane? Do all of the buttons, joysticks, and triggers act as expected?

### Detailed Explaination of Issue
Replace this text with a detailed explaination of your issue. Simply saying "it doesn't work" is a quick way to ensure that your issue is never fixed. Without detailed information on your specific issue, we cannot fix your issue. Including as much information as possible in this section increases the likelyhood of getting your issue quickly and effectively resolved.

FranticRain, 2018-12-04:
Could you run `ls /Library/Extensions` in your terminal and post the output here? If you see a `360Controller.kext` in there, please also run `sudo kextutil /Library/Extensions/360Controller.kext` and tell me what that says.

L0VEFIRE, 2018-12-04:
I get … 

360Controller.kext		CaptoDevice.kext
ACS6x.kext			EnergyDriver.kext
ATTOCelerityFC8.kext		HighPointIOP.kext
ATTOExpressSASHBA2.kext		HighPointRR.kext
ATTOExpressSASRAID2.kext	PromiseSTEX.kext
ArcMSR.kext			SoftRAID.kext
CalDigitHDProDrv.kext

Then … 

Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

> On Dec 4, 2018, at 11:40 AM, Drew Mills <notifications@github.com> wrote:
> 
> /Library/Extensions

FranticRain, 2018-12-04:
So the driver was correctly installed, but the Preference Pane in missing. You can either attempt a re-install to add it, or I can guide you through a manual uninstall if you'd prefer.

L0VEFIRE, 2018-12-04:
Already been down the reattempt route, so let try the manual uninstall method see how that works. Thanks

> On Dec 4, 2018, at 11:40 AM, Drew Mills <notifications@github.com> wrote:
> 
> sudo kextutil /Library/Extensions/360Controller.kext

FranticRain, 2018-12-04:
Open your terminal and run:
```
curl -o delete.sh https://raw.githubusercontent.com/360Controller/360Controller/875b4ca7bcf4c83aab2a6c290a8a411ac2e136d8/Install360Controller/Scripts/upgrade.sh &&
sudo chmod u+x delete.sh &&
sudo ./delete.sh && rm delete.sh
```
And enter your password as required

FranticRain, 2019-04-06:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-04-06 by FranticRain