360Controller/360Controller #811: Xbox 360 controller doesn't show up in 0.16.10, does show up in 0.16.9

Dave-9999, 2018-12-05:
### Type of Controller
Xbox 360 with wireless PC USB receiver

### OS Version
10.14.1

### Driver Version
0.16.9, 0.16.10

### Connection Method
Wireless Adapter

### Device Name and Info

>Device Name: Wireless Xbox 360 receiver
<br/>Product ID:
<br/>Vendor ID: Microsoft

### Response of Controller
Controller is not found in 0.16.10, works fine in 0.16.9

### Response in Preference Pane
Controller is not found in 0.16.10, does not show up  in 0.16.9 Preference Pane

### Detailed Explaination of Issue
As above

mele113, 2018-12-05:
I run 10.13.6 High Sierra and as far as I understand I can only use "v 0.16.5 Controller" (anything later doesn't support wireless adapter... in v016.6 "The wireless controller driver has been removed to fix issues with macOS 10.11 and later.").........releases found here:     https://github.com/360Controller/360Controller/releases

Dave-9999, 2018-12-05:
Just to clarify, I'm using the Xbox 360 controller connected to a MICROSOFT Xbox 360 Wireless Gaming Receiver for Windows, which will appear to the Mac as a USB device, not a BT one. This is supported, and works great in 0.16.9.

https://github.com/360Controller/360Controller#wireless-xbox-360-controllers
"Remember that wireless controllers must be connected using a wireless adapter. "

FranticRain, 2018-12-06:
Pushed a change to the README that improves the visibility of the fact that Wireless Xbox 360 Controllers are no longer supported. The wireless controller not working in 0.16.10 is expected behavior. Before 0.16.10, the wireless driver was included with a wink and a nod that it was unsupported, but you could use it if you wanted to. Too many people ignored this warning and still complained about issues with wireless controllers, so it was fully removed in 0.16.10.

closed 2018-12-06 by FranticRain

Dave-9999, 2018-12-06:
Thanks a million for clarifying, and for your hard work !!!!!