360Controller/360Controller #814: PowerA - Xbox One Enhanced wired controller (3rd party) not recognized

izzybe, 2018-12-17:
### Type of Controller
PowerA - Xbox One Enhanced wired controller (3rd party)

### OS Version
10.14.1

### Driver Version
Version 0.16.10 and Version 0.16.09

## Connection Method
 Wired

### Device Name and Info

>Device Name:
<br/>Product ID: 0x581a
<br/>Vendor ID: 0x24c6

### Response of Controller
Button does not light up, driver mentions no devices found. Even followed several ways of trying to recognize controller. USB is recognized in Mac (USB - system information
).
 In Windows virtual machine, button lights up and controller works.

### Response in Preference Pane
N/A

### Detailed Explaination of Issue
Add 3rd party controller to driver installer.

rjriel, 2018-12-29:
Duplicate of #757

FranticRain, 2019-01-02:
As copied from #757: PDP and PowerA Xbox One controllers are incompatible with macOS 10.11+ thanks to a rewrite of the macOS USB kernel. We cannot resolve this issue. It is a bug in Apple's core OS code. There are currently no plans to proceed with a rewrite of the driver.

closed 2019-01-02 by FranticRain