360Controller/360Controller #815: No devices found - 0.16.10 - Macbook Pro 2017 Touch Bar

HrOvergaard, 2018-12-22:
Xbox 360

Macbook Pro 2017 touchbar - MacOS Mojave - 10.14.2

0.16.10

Wired Wireless controler // Apple USB to Type C adapter

>Device Name:
<br/>Product ID: 0x028f
<br/>Vendor ID: 0x045e 

Response of Controller
None

Response in Preference Pane
None

Detailed Explaination of Issue
None responsing -  Installing and uninstalling different versions of the software, but no luck
Just showing "No devices found"

[image]

[image]

RobertAtPebblebee, 2018-12-22:
seems like the real bug is that the latest driver doesn't work with the latest version of mojave.  can anyone confirm?

midkir, 2018-12-24:
I have the same situation.
Installing the driver on older macOS version, and upgrading to Mojave works fine.
In Mojave, I did not get "System Extension Blocked" dialog in the installation process, so I cannot permit the privilege to the driver.

[image]

iikas, 2018-12-26:
Hello, I actually have macOs mojave and the current version 0.16.10 of the driver for xbox 360 does not recognize the command once connected, is there any way to solve this problem? Please T.T

gerardo8a, 2018-12-28:
I'm facing the same issue, but I got asked the very first time I installed the software for the developer allow access. But after that I can't still use the controller. 
I have a Xbox One controller but the same bug.

I tried removing and reinstalling, also restarting the system but no luck.

HrOvergaard, 2018-12-28:
I'm going to try with a hard wired usb controller for Xbox 360 and also with the microsoft wireless controll reciver (together with the 0.16.5 software of cause) If that does not work I will try the same on an old Mac with Snow Leopard.

If nothing works... I will try with... windows....

theJeej, 2019-01-03:
This problem is persevering, as I have the latest update both for mac and for the software.

yowzadave, 2019-01-03:
I'm experiencing the same issue, on Mojave 10.14.2.

FranticRain, 2019-01-06:
Okay. First things first, the driver still works on macOS Mojave 10.14.2. I just tested it.

@HrOvergaard @midkir @theJeej @yowzadave 
Please ensure that the driver has been allowed in System Preferences -> Security and Privacy -> General. This approval UI is only present in the Security & Privacy preferences pane for 30 minutes after the alert. Until you approve the kext, future load attempts will cause the approval UI to reappear but will not trigger another user alert.

[image]

@iikas Make a new issue. Make sure to mention in your new issue whether or not the preference pane says "No devices found" or if a device is actually found

@gerardo8a That looks like a successful load. Very different from these issues. Please make a new issue.

@Poisonousbeef Wireless controllers are not currently supported by the driver. For further information, please reference the README.

If I didn't call you out and you still need help, make a new issue.

RobertAtPebblebee, 2019-01-07:
I am not getting this warning in Security preferences.  Though I am running 10.14.3.  The 'driver enabled' checkbox is never checked, and every time I check the box authorize the change and leave the preference and come back to it, the check box is yet again not checked (and still no feedback in the security preference).  Not sure if it matters but I have SiP disabled.  I tried building driver myself but fails to build with missing IOKit header file.  Looks like the latest code is referencing an older version of IOKit that may no longer have the device.h or interface.h files.  I can open a fresh issue I guess.

FranticRain, 2019-01-07:
@RobertAtPebblebee The driver enabled checkbox has nothing to do with your issue. It's just some weird behavior that is working as expected, even when the driver is working correctly. If SiP is disabled, the option won't appear, because the driver should work automatically. The reason you can't build is because the driver has to be built against Xcode 6.4. An upgrade to modern IOKit is hindered by a memory leak in macOS. Make a new issue, filling out the template, and include the result of `kexutil /Library/Extensions/360Controller.kext` in your details.

iikas, 2019-01-07:
SOLVED
Effectively, installing the most recent version of the driver 360 in macOs Mojave version 0.16.10 did not work even though it allowed the option in the preferences / security panel, but after an exhaustive search it managed to work with the driver 360 version 0.16.5, same way accept the permission in the preferences panel and go!
😄

iikas, 2019-01-07:
SOLVED
Effectively, installing the most recent version of the driver 360 in (macOs Mojave version 10.14) did not work even though it allowed the option in the preferences / security panel, but after an exhaustive search it managed to work with the driver 360 version 0.16.5, same way accept the permission in the preferences panel and go!
😄 I hope you also work for you

HrOvergaard, 2019-01-07:
Just got it working with 0.16.5 together with the "PC wireless gaming receiver" and xbox 360 wireless controller. GAME ON

FranticRain, 2019-01-07:
As mentioned in the README under the wireless 360 controller section, since it seems like you two are trying to use wireless 360 controllers: https://github.com/360Controller/360Controller#wireless-xbox-360-controllers
>As of macOS 10.11, Wireless Xbox 360 controller support causes kernel panics. This issue cannot be resolved with minor changes to the driver, and requires that the driver be re-written from scratch to resolve the issue. Due to an excess of caution, we have disabled Wireless Xbox 360 controller support as of 0.16.6. If you want to use a wireless controller, download 0.16.5 or earlier and disable the driver before the computer enters a "sleep" state in order to prevent kernel panics. Alternatively, you can revert to a macOS version before 10.11 to avoid this issue.

closed 2019-01-07 by FranticRain