360Controller/360Controller #816: Afterglow Wired Controller for Xbox One not found

jrogers512, 2018-12-23:
### Type of Controller
Replace this text with: Xbox One

### OS Version
macOS: 10.14.2 (18C54)

### Driver Version
360Controller 0.16.10

### Connection Method
Wired (USB 3.0)

### Device Name and Info
>Device Name: Afterglow Wired Controller for Xbox One
<br/>Product ID: 0x02b8
<br/>Vendor ID: 0x0e6f

### Response of Controller
Lights up

### Response in Preference Pane
"No Devices Found"

### Detailed Explanation of Issue
I appear to have a similar issue as seen in https://github.com/360Controller/360Controller/issues/689 .  

`360Controller $ brew instal lsusb; lsusb | awk '/USB 3.0 Bus/{y=1;next}y'
Warning: lsusb 1.0 is already installed and up-to-date
To reinstall 1.0, run `brew reinstall lsusb`

      Host Controller Driver: AppleUSBXHCIPPT
      PCI Device ID: 0x1e31
      PCI Revision ID: 0x0004
      PCI Vendor ID: 0x8086

        Afterglow Wired Controller for Xbox One:

          Product ID: 0x02b8
          Vendor ID: 0x0e6f
          Version: 1.01
          Serial Number: 00009767FA5A59AF
          Speed: Up to 12 Mb/sec
          Manufacturer: Performance Designed Products
          Location ID: 0x14100000 / 7
          Current Available (mA): 500
          Extra Operating Current (mA): 0
` 

From Console with "ANY:Kernel":
`default	21:55:42.295045 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.295059 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.295064 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.295092 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.295153 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.295191 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.295222 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.295247 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.295287 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.295307 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.295333 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.295384 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.295425 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.295470 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.295506 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.295552 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.295587 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.295620 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.295654 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.295697 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.295724 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
[... 44 lines cut ...]
default	21:55:42.296307 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.296316 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.296325 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.296333 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.296342 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.296351 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.296359 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.296369 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.296377 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.296386 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.296395 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.296403 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.296410 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.296418 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.296423 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.296428 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.296434 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.296439 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.296444 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
default	21:55:42.296449 -0600	kernel	Afterglow Wired Controller for Xbox One: family specific matching fails
`

I've confirmed the product id and vendor id are already in the driver, so should work.  I see the device in system information and also using brew's lsusb (see above).  

Any ideas why the 360Controller prefpane isn't listing the controller?

sideofries, 2018-12-28:
I have an identical setup with the exact same identical problem. The only difference from what jrogers512 found is that my controller does not even light up. Here's how it shows up in the USB device tree:
Afterglow Wired Controller for Xbox One:

  Product ID:	0x02b8
  Vendor ID:	0x0e6f
  Version:	2.13
  Serial Number:	[[redacted]]
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Performance Designed Products
  Location ID:	0x14600000 / 3
  Current Available (mA):	500
  Extra Operating Current (mA):	0

sideofries, 2018-12-29:
Ah, found this: https://github.com/360Controller/360Controller/issues/790
I guess there's no hope for this model.

FranticRain, 2019-01-02:
PDP and PowerA Xbox One controllers are incompatible with macOS 10.11+ thanks to a rewrite of the macOS USB kernel. We cannot resolve this issue. It is a bug in Apple's core OS code. There are currently no plans to proceed with a rewrite of the driver.

closed 2019-01-02 by FranticRain