360Controller/360Controller #819: XBox One controller not found in Preferences

Mirovsky, 2018-12-26:
### Type of Controller
Xbox One

### OS Version
10.14 Mojave

### Driver Version
0.16.10

### Connection Method
Wired

### Device Name and Info
>Device Name: Controller
<br/>Product ID: 0x02ea
<br/>Vendor ID: 0x045e

### Response of Controller
None

### Response in Preference Pane
No devices found

### Detailed Explaination of Issue
- Controller shown in About -> System Report -> USB as Controller
- Allow certificate not shown
- Tried load extension from terminal
- kextload command printed this: 

```
/Library/Extensions/360Controller.kext failed to load -(libkern/kext) authentication failure (file ownership/permissions); check the system/kernel logs for errors or try kextutil(8).
```

- Popup about certificate appeared, clicked on allow, nothing changed
- Output of kextutil:

```
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Kext rejected due to insecure location: <OSKext 0x7fcad9d58550 [0x7fff8755d8e0]> { URL = "file:///Library/StagedExtensions/Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Kext rejected due to insecure location: <OSKext 0x7fcad9d58550 [0x7fff8755d8e0]> { URL = "file:///Library/StagedExtensions/Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Diagnostics for /Library/Extensions/360Controller.kext:
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents
```

FranticRain, 2019-01-02:
The output of kextutil is saying that you haven’t accepted the signing certificate as you’ve said. The prompt should be available in your System Preferences -> Security shortly after plugging in the controller. If you don’t get it, then you may need to re-install the kext to get the prompt. Without it, the kext will not work.

Solced, 2019-01-05:
Hey there! 
I got a similar problem. I downloaded and installed 360Controller version 0.16.10 after i used 360Controller Version 0.11 on MacOS Sierra.
The Xbox One Controller is working but in the preference pane the app is nowhere to be found. The older version was in the preference pane. I tried to do the steps you mentioned above but there is no blocked app from Drew Mills appearing.
The only Data i have is the file in System/Library/Extensions/360Controller.kext
Nothing more i can find. So how do i uninstall the driver when there is just this data to show it?

FranticRain, 2019-01-06:
@Solced Please make new issues for entirely separate issues. Adding on to unrelated issues only hampers my ability to provide effective support. The files you are looking to delete are:
```
/System/Library/Extensions/360Controller.kext
/System/Library/Extensions/Wireless360Controller.kext
/System/Library/Extensions/WirelessGamingReceiver.kext
/Library/Extensions/360Controller.kext
/Library/Extensions/Wireless360Controller.kext
/Library/Extensions/WirelessGamingReceiver.kext
/Library/Extensions/XboxOneBluetooth.kext
```
This list includes every file ever added by any version of the driver. You may not have all of these files.

Stenotti, 2019-02-15:
Hi everyone, did you manage to solve the issue? 
I am facing the same problem, "No device found" in the System Preferences 
[image]
I am able to use the xbox controller in Big Picture mode and in some games (I tried Portal 2), but it's not working on Rocket League. It was working until a month ago.

Thanks!

FranticRain, 2019-02-16:
@Stenotti Make a new issue and follow the template.

FranticRain, 2019-05-26:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-05-26 by FranticRain