360Controller/360Controller #820: one light turns on, and says there are no devices

theJeej, 2019-01-03:
360 Controller

MacOS High Sierra

Version 06.10

Wired Connection

Detailed Explaination of Issue
Whenever I plug the controller in, I have to press the Xbox button. The light goes red, but only on one quarter of the circle. the player one side. then in system preferences, the add-on says that no devices have appeared or have been found. does anybody have a solution?

FranticRain, 2019-01-06:
You must update your original post with your controller information. A guide on how to do so can be found here: https://github.com/360Controller/360Controller#how-do-i-find-my-vendor-id-and-product-id.

Controller information is only not required for issue that do not involve controllers. For example, a feature request that isn't specific to any controller, or maybe a typo somewhere in the code or documentation. For all issues individuals have with getting a controller working, controller information is ABSOLUTELY VITAL to solving your problem. Not only does it let us know if your controller is already supported, it also lets us know that the operating system can see the controller and is a crucial step in resolving your issue.

Please notify my when you have updated your original post with your controller information, or let me know if you are unable to find it.

FranticRain, 2019-04-06:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-04-06 by FranticRain