360Controller/360Controller #822: Razer Wolverine Compatibility Issues

benniejet5, 2019-01-06:
### Type of Controller
Razer Wolverine Ultimate

### OS Version
Sierra 10.12.6

### Driver Version
0.16.10

### Connection Method
Wired

### Device Name and Info

>Device Name: Razer Wolverine Ultimate
<br/>Product ID: 0x0a14
<br/>Vendor ID: 0x1532

### Response of Controller
Will light up and vibrate, but is unrecognized by the computer.

### Response in Preference Pane
Will recognize that a controller is connected, but no changes can be made, and no movement is made by the controller. 

### Detailed Explaination of Issue
Please refer to thread #796 (linked bellow), as my issue is the same. I've read other threads and saw that packet sniffing needed to be done on a PC. I can do this, but need help with the process. Please contact me so we can work together to get this resolved for everyone who has had this issue in the past! Thank you.

https://github.com/360Controller/360Controller/issues/796

FranticRain, 2019-01-06:
The guide for how to sniff packets using a Windows machine is documented here: https://vestigl.com/sniffing/index.html

benniejet5, 2019-01-07:
great, I'll get this done tomorrow and link the results.

FranticRain, 2019-05-26:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-05-26 by FranticRain