360Controller/360Controller #824: Xbox One Controller Not Found

cfrederick95, 2019-01-12:
### Type of Controller
Xbox One
### OS Version
10.14.2 (18C54)
### Driver Version
360Controller 0.16.10
### Connection Method
Wired
### Device Name and Info
Bluetooth USB Host Controller:

  Product ID:	0x8290
  Vendor ID:	0x05ac (Apple Inc.)
  Version:	1.66
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Broadcom Corp.
  Location ID:	0x14300000 / 5
  Current Available (mA):	500
  Current Required (mA):	0
  Extra Operating Current (mA):	0
  Built-In:	Yes

### Response of Controller
When I plug it in, nothing happens. When I press the xbox button on the controller it blinks slowly for a few seconds and then turns off again. 

### Response in Preference Pane
"No device found"

### Detailed Explaination of Issue
I have approved permissions and done a manual uninstall/reinstall of the driver. 

This is what my terminal is showing:
Teds-MacBook-Pro:~ colefrederick$ sudo kextutil /Library/Extensions/360Controller.kext
Password:
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

FranticRain, 2019-01-12:
So that controller information that you have given me is for the Bluetooth module in your computer. Based on your terminal output, it looks like you are getting a successful load of the driver, so you're good to go. So we've got one of two problems: 1. You're using some sort of 3rd party controller, in which case you need to go back to that same menu and find the real controller product id and vendor id. Or 2. The controller information isn't there, because you're using what is often referred to as a "charge" cable. Basically, not all USB cables are made the same, and some of them only have the power lines connected in them. There are no data lines in the cable. This means that the controller can't talk to the computer, and thus the driver can't work. Try a different cable, just to make sure. If you can find the controller in your system information, then the computer can see it.

cfrederick95, 2019-01-16:
It was the USB cable. Thank you for the help! Everything is working now.

closed 2019-01-16 by cfrederick95