360Controller/360Controller #826: Add support for Sony DualShock 4 (PS4)

diedummydie, 2019-01-21:
### Type of Controller
(PlayStation 4) **DualShock 4** 
(Sony official, newest, bundled with PS4 Slim and PS4 Pro)

### OS Version
10.14.1

### Driver Version
0.16.10

### Connection Method
Wired

### Device Name and Info
Device Name On Windows 7: "Wireless Controller" (even when USB-connected)
Device Name On Mac: "Wireless Controller \#1" (second controller lists as "Wireless Controller \#2")
Product ID: 0x09cc
Vendor ID: 0x054c

### Response of Controller
Not recognized by 360Controller
Works with Steam, OpenEmu, Joystick Mapper, and Enjoy2.
(My Xbox 360 wired controller works correctly with 360Controller on this system.)

### Response in Preference Pane
Not recognized by 360Controller
In Enjoy2 buttons are listed like this:
X (Xbox equiv. A) = Button 2
O (Xbox equiv. B) = Button 3
□ (Xbox equiv. X) = Button 1
∆ (Xbox equiv. Y) = Button 4
When using USBPcap in a Windows 7 vm, I pressed the buttons in an order equivalent in layout to what you specify for Xbox controllers — A B X Y, i.e. X (Button 2), O (Button 3), □ (Button 1), ∆ (Button 4). Despite the very brief duration of the captures, the files are ~250kb. I think these have an accelerometer, so that may be the reason.

[DS4-USBPcap.zip](https://github.com/360Controller/360Controller/files/2779992/DS4-USBPcap.zip)

### Detailed Explaination of Issue
I want my DS4 to masquerade as a 360 controller for the sake of massively increased game compatibility. I know PS4 controller compatibility is not the goal of your project, and I understand if its protocol is too different from the ones that do work with 360Controller.

FranticRain, 2019-01-22:
In short: I will not implement this and I will not support anyone in implementing this.

Long answer: There are a couple things about this concept that I definitely disagree with. First, providing a compatibility layer is not the job of a driver. I hate every part of this driver that acts as a compatibility layer. Second, I don't want to support any controller that does not work on an Xbox console. This is because it inspires a little bit of "what-about-ism." If we implement DS4 support, why not DS3 support, why not Nintendo Switch Support, why not support Android game controllers. I support this driver in my free time, for my own entertainment. I do not, under any circumstances want to balloon the complexity and difficulty of support of this driver. If some intrepid person comes along and decides that they want to support more controllers, much like I did with Xbox One controllers, then they can take it.

diedummydie, 2019-01-22:
Understood. Thank you for the clear and thorough explanation.

closed 2019-01-22 by diedummydie