360Controller/360Controller #830: 360Controller Please help me fix

Pinki21, 2019-01-25:
Type of Controller
 Xbox 360

OS Version
10.14

Driver Version
0.16.5

Connection Method
Wired

 Device Name and Info
Device Name:Vendor-Specific Device
Product ID:0x028f
Vendor ID:0x045e (Microsoft Corporation)

Response of Controller
Ok so the controller just blinks when i press the button it blinks to infinite nothing just blinks the full ring blinks

Response in Preference Pane
No device found and i activated the allow thingy in the security & privacy

Detailed Explaination of Issue
so i Tried TattieBogle doesnt work tried yours doesnt work tried repluging the usb in controller 
tried to restart computer tried to hold the button on the controller for more than 10sec idk why xd

FranticRain, 2019-01-25:
I need you to open up your terminal and run `sudo kextutil /Library/Extensions/360Controller.kext` and tell me what it says.

Pinki21, 2019-01-25:
> I need you to open up your terminal and run `sudo kextutil /Library/Extensions/360Controller.kext` and tell me what it says.
it Says
sudo kextutil /Library/Extensions/360Controller.kext
Password:
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

FranticRain, 2019-01-25:
Okay, so you've successfully set up the driver and your computer recognizes your controller. What's concerning to me is that you shouldn't have to press the guide button to turn on a wired controller. The computer should always turn it on. I just want to make sure, are you using an originally wired controller, or a controller connected with a wireless-to-wired adapter called the "Play and Charge Kit?" Because by all accounts, this should work. But either you're using a play and charge kit, or you have a bad controller.

Pinki21, 2019-01-25:
> Okay, so you've successfully set up the driver and your computer recognizes your controller. What's concerning to me is that you shouldn't have to press the guide button to turn on a wired [...]

So everytime i plug the usb in the computer and replug the light on the wire turns red then  back to yellow but the button stays not blinking and what you mean play and charge kit i have this controller from xbox360 that i had

Pinki21, 2019-01-25:
so ?

Pinki21, 2019-01-25:
SO WHAT NOW ?

FranticRain, 2019-01-25:
Bud, I have a full time job so I can't immediately solve all of your problems. I'll get to them when I have time. The play and charge kit doesn't turn a wireless Xbox 360 Controller into a wired one. All that cable does is provide power to the controller. This means that no data is being transferred to the PC, and therefore it can't reach the driver. Because of this, this ISN'T a wired controller. It is still a wireless controller. At this time, wireless 360 controllers are not supported by this driver.

closed 2019-01-25 by FranticRain