360Controller/360Controller #831: Xbox One

weejkyjack, 2019-01-26:
### Type of Controller
Replace this text with: Original Xbox, Xbox 360, Xbox One, All or N/A

### OS Version
Replace this text with the version of macOS that you are running

### Driver Version
Replate this text with the version or versions of the driver that you are using or have tested

### Connection Method
Replace this text with: Wired, Wireless Adapter, Bluetooth, N/A or more specific method if applicable

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID:
<br/>Vendor ID:

### Response of Controller
Replace this text with information on how the physical controller reacts. Does the guide button light up? Does it vibrate? Etc.

### Response in Preference Pane
Replace this text with information about the controller's reaction in the preference pane. Does the controller appear in the preference pane? Do all of the buttons, joysticks, and triggers act as expected?

### Detailed Explaination of Issue
Replace this text with a detailed explaination of your issue. Simply saying "it doesn't work" is a quick way to ensure that your issue is never fixed. Without detailed information on your specific issue, we cannot fix your issue. Including as much information as possible in this section increases the likelyhood of getting your issue quickly and effectively resolved.

FranticRain, 2019-01-27:
You must update your original post with your controller information. A guide on how to do so can be found here: https://github.com/360Controller/360Controller#how-do-i-find-my-vendor-id-and-product-id.

Controller information is only not required for issue that do not involve controllers. For example, a feature request that isn't specific to any controller, or maybe a typo somewhere in the code or documentation. For all issues individuals have with getting a controller working, controller information is ABSOLUTELY VITAL to solving your problem. Not only does it let us know if your controller is already supported, it also lets us know that the operating system can see the controller and is a crucial step in resolving your issue.

Please notify my when you have updated your original post with your controller information, or let me know if you are unable to find it.

FranticRain, 2019-05-26:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-05-26 by FranticRain