360Controller/360Controller #833: xbox 360 wired | No device found

egorcherkasov, 2019-02-02:
### Type of Controller
Xbox 360

### OS Version
10.14.2

### Driver Version
0.16.10

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Controller
<br/>Product ID: 0x028e
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
When the controller is connected, the light ring lights up and goes out after a second. The program displays "No devices found".

### Response in Preference Pane
"No devices found". The program does not respond to the controller connection.

### Detailed Explaination of Issue
Hi! The driver is allowed in System Preferences -> Security and Privacy -> General. Controller is work on PC machine. iMac (Retina 4K, 21.5-inch, 2017). I used all USB ports and reboot with the connected controller. 

The result of launching: 
`sudo kextutil /Library/Extensions/360Controller.kext`

Notice: /Library/Extensions/360Controller.kext has debug properties set.
Kext rejected due to system policy: <OSKext 0x7ff0cbd7bec0 [0x7fff916978f0]> { URL = "file:///Library/StagedExtensions/Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Kext rejected due to system policy: <OSKext 0x7ff0cbd7bec0 [0x7fff916978f0]> { URL = "file:///Library/StagedExtensions/Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Diagnostics for /Library/Extensions/360Controller.kext:
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

Please, help! And thank you for a great work!)

FranticRain, 2019-02-03:
Please ensure that the driver has been allowed in System Preferences -> Security and Privacy -> General. This approval UI prompt is only present in the Security & Privacy preferences pane for 30 minutes after an alert, such as the one that appeared during installation. Until you approve the kext, future load attempts will cause the approval UI to reappear but will not trigger another user alert. You can trigger a kext load by either re-connecting the controller or by running

```sudo kextutil /Library/Extensions/360Controller.kext```

The approval UI is as follows:

[image]

gllmt, 2019-04-21:
Hello, 
I have a same issue, on macBook pro 2016 (mojave 10.14.4) with Xbox360 controller usb...
Did you find a solution

thank you, for your work !

FranticRain, 2019-04-21:
@gllmt The solution to the problem posed in this issue is as listed. If that doesn't solve your problem, make a new issue following the issue template.

closed 2019-04-21 by FranticRain