360Controller/360Controller #834: MacBook Pro won't boot up.

blainedont, 2019-02-04:
I have Mojave 10.14.2

I installed your app to make a Xbox controller work.

https://github.com/360Controller/360Controller/releases

Now the MacBook won't boot up after the suggested restart.

FranticRain, 2019-02-04:
Hold Command + V when you boot the machine. When the text stops moving, take a picture of your screen. That will help us fix this issue more permanently.

Then, in order to fix your machine, you will need to shut it down again, and then hold Command + R to get into recovery mode. From there, open the terminal and run these commands:

```
rm /System/Library/Extensions/360Controller.kext
rm /System/Library/Extensions/Wireless360Controller.kext
rm /System/Library/Extensions/WirelessGamingReceiver.kext
rm /Library/Extensions/360Controller.kext
rm /Library/Extensions/Wireless360Controller.kext
rm /Library/Extensions/WirelessGamingReceiver.kext
rm /Library/Extensions/XboxOneBluetooth.kext
```

You may not have all of these files. This is just the full list to be safe. And one final question, do you know which version of the driver you installed?

blainedont, 2019-02-04:
It was 0.16.10.

I ended up reinstalling Mojave on the MacBook but I am hesitant to do this
again.  Ultimately I want to use the xbox controller on my mac mini.  Any
tips you recommend?

Richard

FranticRain, 2019-02-04:
Without your Cmd + V trace, it is impossible to know why you experienced this problem. Without it, I can't promise you anything.

closed 2019-02-04 by FranticRain

blainedont, 2019-02-04:
I will attempt the mac mini at some point.  I just need to know if you have
had successful attempts with Mojave.

Richard

blainedont, 2019-02-04:
> I've run the driver on 10.14, 10.14.1, 10.14.2, and 10.14.3 with success.

Thanks!