360Controller/360Controller #835: Macbook Pro Hangs on Boot

duno9, 2019-02-04:
### Type of Controller
Xbox One

### OS Version
10.13 High Sierra

### Driver Version
0.16.10

### Connection Method
N/A

### Device Name and Info
N/A

### Response of Controller
N/A

### Response in Preference Pane
N/A

### Detailed Explaination of Issue
My computer hangs on boot. I have tried using recovery mode to repair the disk, it said the disk was fine. I looked through my library extensions and system library extensions and found none of the .kext files that have been listed in a previous issue (360controller.kext, wirelessgamingreceiver.kext, etc..). I performed a manual boot from another hang on boot issue posted here 2 months ago, but it gave me the message "could not resolve host raw.githubusercontent.com". I have the verbose mode text in a picture from when the boot hangs, apologies if it is hard to read. 

[image]

[image]

Please help, I need my computer back.

FranticRain, 2019-02-04:
If you have successfully removed the files from
```
/System/Library/Extensions/360Controller.kext
/System/Library/Extensions/Wireless360Controller.kext
/System/Library/Extensions/WirelessGamingReceiver.kext
/Library/Extensions/360Controller.kext
/Library/Extensions/Wireless360Controller.kext
/Library/Extensions/WirelessGamingReceiver.kext
/Library/Extensions/XboxOneBluetooth.kext
```
Then the driver is not installed on your machine and isn't causing your problem. Those are the only executed files contained as part of the driver. If it isn't installed, it isn't doing anything to your machine.

johnmcollier, 2019-02-21:
Yeah, just hit this exact same issue too - Couldn't boot into safe mode either, and FileVault would hang during decryption. Only solution was to wipe the disk and restore from a backup

FranticRain, 2019-02-21:
This is a known issue. It seems to be caused by some sort of incompatibility between the shim written for the old versions of the IOKit used for this driver and newer versions of IOKit developed in macOS 10.11. For more discussion of this issue and an explanation as to why we haven't upgraded yet, see #698

closed 2019-02-21 by FranticRain