360Controller/360Controller #837: Wired XBox360 controller disconnecting while vibrating

Frinkey, 2019-02-07:
### Type of Controller
Xbox 360

### OS Version
macOS 10.14.2

### Driver Version
0.16.10

### Connection Method
Wired via apple usb-c (usb 3.1 bus) to usb-c/hdmi/usb-a adapter (usb 2.0)

### Device Name and Info
>Device Name: Controller
<br/>Product ID:	0x028e
Vendor ID:	0x045e  (Microsoft Corporation)
Version:	1.14
Serial Number:	05813A7
Speed:	Up to 12Mb/sec
Manufacturer:	Holtek
Location ID:	0x14110000 / 26
Current Available (mA):	500
Current Required (mA):	500
Extra Operating Current (mA):	0

### Response in Preference Pane
All joysticks, buttons etc works fine

### Detailed Explaination of Issue
the controller simply disconnect for a moment and then reconnects while vibrating. In the driver panel, when the trigger is clamped, the controller begins to vibrate and periodically disconnects. I do not have a possibility to test controller on another computer or console. Is it a bug of driver or broken controller?

FranticRain, 2019-02-07:
Without any way to test it on another machine, it is more likely that this is in some way tied to a loose connection in the chain that gets disconnected during rumble. It is difficult to imagine a situation in which this is a driver bug that doesn't cause a kernel panic.

Frinkey, 2019-02-08:
Controller works fine at another windows PC, I plugged it in and started the vibration for two minutes and it didn't stop. I think there is a problem with usb-c adapter... The experiments continue =)

Frinkey, 2019-02-09:
Tried with another usb-c -> usb-a adapter and gamepad works fine... =( I am disappointed with the original adapter

closed 2019-02-09 by Frinkey

FranticRain, 2019-02-09:
Best guess is that the original adapter wasn't capable of providing enough current to the Xbox One controller to drive the motors for rumble.

Frinkey, 2019-02-09:
> Best guess is that the original adapter wasn't capable of providing enough current to the Xbox One controller to drive the motors for rumble.

[image]
500 mA

But now I want to know, why my usb 3 adapter gives me so little current

FranticRain, 2019-02-09:
That's the normal amount of current available to a USB port. That number is how much the computer is capable of providing to it. Not how much the device can actually handle. You'd have to tear down the dongle, and probably have a degree in electrical engineering to find out why it doesn't work right.