360Controller/360Controller #838: Lost connection disables spoof.

itskando, 2019-02-16:
### Type of Controller
Xbox One Wireless

### OS Version
10.13.6

### Driver Version
0.16.10

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Controller
<br/>Product ID: 0x02ea
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
Lights up.
Vibrates.

### Response in Preference Pane
Responds.

### Detailed Explaination of Issue
I have brand new controllers and brand new cables (brand: Monoprice).
Even so, when using a wired connection, 
the wire slips, temporarily severing the connection.

When spoofing the Xbox One Wireless Controller 
(to be an Xbox 360 controller),
a slipped connection immediately changes it back to 
an Xbox One Wireless controller.
(Likely seen as a new controller..)

Also, as a separate but related issue, 
the spoof enable switch remains checked, 
despite spoofing no longer being enabled.
[An easy fix to this is to disable then re-enable spoofing.]

The latter is a mild inconveniece; however,
losing the spoof instantly within a game can be very problematic.
Any known workarounds to this issue?

itskando, 2019-02-16:
It might be helpful to allow a spoof all controllers mode,
so that controllers that lose connection instantly go back to 360 when reconnected,
and plugging in 4 controllers all go to 360 controllers.

I think the need for this is more common than the need for
playing with some Xbox One as 360 controllers and some Xbox One as Xbox One controllers.

FranticRain, 2019-02-18:
This is, in some ways, a duplicate of known issue #697. The trick is that drivers were never meant to have "settings," so the way we have implemented setting is hacky at best. Simply put, a controller won't pick up any settings when it is first connected, for some odd reason. That covers your first issue. As for your second issue, that is working as intended. The actual settings database still has the "Pretend to be a 360 Controller" bit of data set to true, so the Preference Pane is doing as it is told. Not really a bug, but rather a weird symptom of the issue. As for your "spoof all controllers" proposal, it wouldn't work for the same reason that the original setting doesn't work.

itskando, 2019-02-18:
Yeah, I figured if it didn't exist yet, 
there was probably a reason along those lines.
Thanks for acknowledging and for all the continuing efforts.