360Controller/360Controller #843: Please add this to the list of Third Party Controllers

dmhuyoung, 2019-02-25:
### Type of Controller
Xbox 360 USB wired controller

### OS Version
MacOS 10.14.3

### Driver Version
0.16.10

### Connection Method
Wired USB

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Inno GamePad (Mad Catz, Inc.)
<br/>Product ID: 0xf401
<br/>Vendor ID: 0x0738

### Response of Controller
The guide button only blinks, no combination of button presses elicits a change in the blinking guide button.

### Response in Preference Pane
I only get "No devices found" in the dropdown menu. No response whatsoever to controller inputs in the preference pane. I often have to re-check the "Enable Driver" box under the "Advanced" tab, but still no response.

### Detailed Explaination of Issue
My wired USB controller is branded as a Microsoft Xbox 360 controller, though MacOS System Information shows it as an Inno GamePad, possibly made by MadCatz, Inc. I'm guessing that this particular Vendor and Product ID just hasn't made into the compiled listing of third party controllers for this driver. Could you please add it to the next updated driver listing? I looked into adding it myself, but quickly got in over my head. Thanks!

dmhuyoung, 2019-02-25:
All four parts of the green circle around the "Xbox" guide button blink simultaneously.

FranticRain, 2019-03-04:
This device is included in the newly released 0.16.11.

closed 2019-03-04 by FranticRain