360Controller/360Controller #846: afterglow wired controller not recognized

jcb55, 2019-03-02:
### Type of Controller
Afterglow gamepad

### OS Version
Mojave 10.14.2

### Driver Version
Latest

### Connection Method
Wired

### Device Name and Info
Afterglow Gamepad for Xbox 360:

  Product ID:	0x0113
  Vendor ID:	0x0e6f
  Version:	1.02
  Serial Number:	00005240
  Speed:	Up to 12Mb/sec
  Manufacturer:	Performance Designed Products
  Location ID:	0x1a120000 / 2
  Current Available (mA):	500
  Current Required (mA):	500
  Extra Operating Current (mA):	0

## Response of Controller
Guide button lights no

### Response in Preference Pane
Controller does not appear 

### Detailed Explanation of Issue
I think Afterglow has many device/vendor id's

FranticRain, 2019-03-04:
This device is included in the newly released version 0.16.11.

closed 2019-03-04 by FranticRain

jcb55, 2019-03-05:
Yeah seems to work, thanks