360Controller/360Controller #850: Controller not found

P-UPPY, 2019-03-07:
### Type of Controller
Xbox One Controller (w/ bluetooth support)

### OS Version
10.14.1

### Driver Version
0.16.11

### Connection Method
Wired via USB-A cable. 

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID: 0x8290
<br/>Vendor ID: 0x05ac (Apple Inc.)

### Response of Controller
Vibrates upon plugging into Mac, does not light up. 

### Response in Preference Pane
Does not show controller in dropdown menu, the 'Activate Driver' is also unticked. Turning on the driver does not change anything.

### Detailed Explaination of Issue
I have been trying for the last few hours to get my controller to work. I've installed, uninstalled, reinstalled with no change. 
After searching the internet, I found a suggestion to run this command:
**sudo kextutil /Library/Extensions/360Controller.kext**
Doing so yields this response:
Warnings:
**Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done):
ChatPadKeyboardEvents**

I've tried turning my controller on, connecting to my Mac via bluetooth before hooking it up with the usb-a cable and vice versa. Any help would be appreciated, thanks! 👍

FranticRain, 2019-03-08:
So based off of what you've told me, I can tell that the driver is working just fine. But what I can tell you is that that product and vendor combination is not your controller. It's a little obvious because I'm pretty sure that Apple didn't make your controller! Now this means that you either just copied the wrong device, or more likely, your computer doesn't see your controller. This is likely caused by using a "charge" cable. These are cables that only have connections for power in them, and no connections for data. This means that macOS can't see your controller, and therefore the driver can't see your controller. Try a different cable and see if you can find your actual vendor and product ID.

FranticRain, 2019-05-26:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-05-26 by FranticRain