360Controller/360Controller #851: Driver won't recognize original Xbox controller but USB does

TestyTestface, 2019-03-18:
### Type of Controller
Original Xbox

### OS Version
10.13.6

### Driver Version
0.16.11

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: N/A
<br/>Product ID: 0x0288
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
The controller does not react.

### Response in Preference Pane
The controller does not show up in the preference pane.

### Detailed Explaination of Issue
I have enabled the driver in the Security & Privacy tab.  After plugging in the controller I find that the driver does not see it for some reason.  It worked after a few minutes yesterday...no idea how it came to recognize the controller because to my knowledge nothing changed.  Today I am unable to get the driver to recognize the controller.  I've restarted and uninstalled to no avail.  Thanks for your help!

TestyTestface, 2019-03-19:
Some more detail: I notice that the driver only picks up the remote when I reinstall the driver and the remote is plugged in.  Once I unplug and replug the remote, the driver fails to find it.

FranticRain, 2019-03-20:
Tagging @kasbert, as he is the implementer of original Xbox driver.

TestyTestface, 2019-03-29:
sup, kas.  correction to above: the driver works when i restart my machine, but if i unplug and replug the controller without restarting the driver fails to pick up the controller.

TestyTestface, 2019-06-15:
FYI, 1.0.0-alpha.3 resolves this issue for me (though I sometimes have to unplug and replug the controller to get it to recognize, small inconvenience next to a restart).  Thank you, all!

closed 2019-06-15 by TestyTestface

friguron, 2019-06-15:
Will try it also, as I was in the same situation. I'll report any new finding if ever.

FranticRain, 2019-06-15:
Fair warning, I have absolutely no idea how this got fixed. I didn't touch any of the core original Xbox code. It could have been something messed up in the shared code that it didn't like? I don't know for sure. Would be interesting to know which alpha it was first fixed in. Because I really doubt it was fixed in just alpha3. I basically touched none of the wired driver code in that release vs alpha2.