360Controller/360Controller #852: HORI Real Arcade Pro.N Hayabusa works if plugged in on boot; not otherwise

sypl, 2019-03-18:
### Type of Controller
HORI Real Arcade Pro.N Hayabusa arcade stick

### OS Version
10.14.3 Mojave

### Driver Version
0.16.11

### Connection Method
Wired

### Device Name and Info
Device Name: RAP.N4
Product ID: 0x00ae
Vendor ID: 0x0f0d  (HORI CO., LTD.)

### Response of Controller
Works perfectly if stick is plugged in when I boot up machine. 

If I plug it in after boot, or pull it out and then plug in again: There are just two LEDs on this fight stick, turbo and assign. The turbo lights up briefly if I have the switch set to on when I plug it in. It then flickers on and off, consistent with the OS constantly recognising and unrecognising the device, which is what system report shows. 

### Response in Preference Pane
Fine if plugged in on boot. Shows as xbox controller.

If I plug it in after boot, or pull it out and then plug in again: Shows "No devices found"

### Detailed Explanation of Issue
If I have the stick plugged in when I boot up the machine then this driver works like a charm. The device is listed in the plist, so picks up fine. In system report it shows under USB devices as "RAP.N4".

If I then unplug the device and re-plug it in, it no longer shows in devices of the preferences pane for this driver. Looking in system report and refreshing through USB device tree, the RAP seems to be flickering on and off, as if OS is recognising it one second, and then not the other. Repeat ad infinitum. This driver though never picks up on the device again. 

Something else to note: This stick is originally for PS3/PS4/PC(XInput) and is the device "HoriRAPN4" in the plist.

FranticRain, 2019-03-20:
This is going to be a very tough one. Simply put, there is nothing we do differently between on-boot and after-start. This is likely going to be an issue with macOS that we can't solve. I just want to confirm some things real quick, though.

The preference pane doesn't recognize new devices that are connected while it is open. If you re-connect the controller, close the preference pane, and then re-open the preference pane, the controller still ins't recognized, correct?

sypl, 2019-03-22:
Yes, that's correct. It never re-appears. And I think you're right, it seems a deeper underlying OS issue. The device manage showing it constantly connecting and reconnecting is the real issue, and that happens with or without this software. Like I said, it does work on reboot, so I at least have that method which I'm grateful for.

Thanks for your help.

closed 2019-10-24 by FranticRain