360Controller/360Controller #853: Driver will not recognize controller.

GeneralSpills, 2019-03-20:
### Type of Controller
PDP Stealth Series Violet Spectral

### OS Version
10.13.6

### Driver Version
0.16.11

### Connection Method
Wired

### Device Name and Info

PDP Wired Controller for Xbox One - Stealth Series | Specter Violet:

  Product ID:	0x02cb
  Vendor ID:	0x0e6f
  Version:	1.11
  Serial Number:	0000FABFFB63EDB9
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Performance Designed Products
  Location ID:	0x14200000 / 9
  Current Available (mA):	500
  Extra Operating Current (mA):	0

### Response of Controller
Nothing at all.

### Response in Preference Pane
Nothing at all.

### Detailed Explaination of Issue
I have followed the install exactly but the controller is completely unresponsive.

FranticRain, 2019-03-20:
Here's a build with your device added.

[Install360Controller.pkg.zip](https://github.com/360Controller/360Controller/files/2986248/Install360Controller.pkg.zip)

closed 2019-03-20 by FranticRain

GeneralSpills, 2019-03-20:
Thank you for your quick reply and new build.