360Controller/360Controller #854: Dont work on Guitar hero dongle from xbox360

yaroslavtoretto, 2019-03-20:
### Type of Controller
Replace this text with: Original Xbox, Xbox 360, Xbox One, All or N/A

### OS Version
Replace this text with the version of macOS that you are running

### Driver Version
Replate this text with the version or versions of the driver that you are using or have tested

### Connection Method
Replace this text with: Wired, Wireless Adapter, Bluetooth, N/A or more specific method if applicable

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID:
<br/>Vendor ID:

### Response of Controller
Replace this text with information on how the physical controller reacts. Does the guide button light up? Does it vibrate? Etc.

### Response in Preference Pane
Replace this text with information about the controller's reaction in the preference pane. Does the controller appear in the preference pane? Do all of the buttons, joysticks, and triggers act as expected?

### Detailed Explaination of Issue
Replace this text with a detailed explaination of your issue. Simply saying "it doesn't work" is a quick way to ensure that your issue is never fixed. Without detailed information on your specific issue, we cannot fix your issue. Including as much information as possible in this section increases the likelyhood of getting your issue quickly and effectively resolved.

yaroslavtoretto, 2019-03-20:
Greetings, add drivers for the device Guitar hero dongle wireless controller. It is now detected, the indicator is on, but the device search button does not respond to it. The same problem was on the drivers on Windows 7 when you apply to this device. On Windows 10 with built-in xbox 360 joystick devices this problem is fixed when these drivers are applied to this device. All you need to do is insert a diver with windows 10 from the joystick 360 into this package and register it with this device. Thanks for understanding. Just want to play Clone Hero on Mac os x)

FranticRain, 2019-03-20:
Wireless controllers have not been supported since macOS 10.11, as noted in the README. Apple made a change to their operating system that broke support for that device type.

closed 2019-03-20 by FranticRain