360Controller/360Controller #855: Fighting Commander for PlayStation 4 & 3

ipdxl, 2019-03-24:
### Type of Controller
[Fighting Commander for PlayStation 4 & 3
](http://stores.horiusa.com/fighting-commander-for-playstation-4-3-officially-licensed-by-sony-playstation-4/)

### OS Version
10.14.3 (18D109

### Driver Version
0.16.11

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Fighting Commander:
<br/>Product ID: 0x0086
<br/>Vendor ID: 0x0f0d  (HORI CO., LTD.)

### Response of Controller
The controller doesn't react to anything, the turbo indicator doesn't turn on

### Response in Preference Pane
Doesn't appear in preference panel

### Detailed Explaination of Issue
it doesn't work

FranticRain, 2019-03-28:
I have absolutely no idea if this will work or not, but here's a build with the device added.

[Install360Controller.pkg.zip](https://github.com/360Controller/360Controller/files/3015956/Install360Controller.pkg.zip)

closed 2019-03-28 by FranticRain

ipdxl, 2019-03-28:
@FranticRain Ok, I'll check.

ipdxl, 2019-03-28:
@FranticRain mmmmm no. It doesn't work.

FranticRain, 2019-03-28:
Then your device simply doesn't work with this driver. We only support devices that work with official Xbox consoles. If your device doesn't work with a console in that series or happen to comply with the standards of that series, then we cannot support it.