360Controller/360Controller #856: Mcbazel Playstation to 360 Converter doesn't work

SpeedySPCFan, 2019-03-30:
### Type of Controller
Mcbazel PS PS2 to Xbox 360 PC Controller

### OS Version
macOS Mojave 10.14.4

### Driver Version
0.16.11

### Connection Method
Wired

### Device Name and Info
>Device Name:
<br/>Product ID: 0x028e
<br/>Vendor ID: 0x0e6f

### Response of Controller
The four lights come on, and then nothing.

### Response in Preference Pane
Controller isn't recognized, no response.

### Detailed Explaination of Issue
I plugged the adapter into both a USB 2.0 hub and directly into the USB 3.0 on my MacBook Pro Retina 15' mid-2014, and got the same response from both: all four lights turn on briefly, then turn off, and the controller is recognized by my MBP but not by the driver. This adapter works fine on Windows 7 and on my Xbox 360.

FranticRain, 2019-04-06:
Let me know if this build of the driver works with the adapter.

[Install360Controller.pkg.zip](https://github.com/360Controller/360Controller/files/3050129/Install360Controller.pkg.zip)

SpeedySPCFan, 2019-04-07:
Yes, that's perfect! Thank you so much! Now I can finally play UNIST in WINE with my arcade stick. Thank you again :)

closed 2019-04-07 by FranticRain