360Controller/360Controller #857: Buttons are not registering when I press them on occasion while in game.

L0VEFIRE, 2019-04-04:
### Type of Controller
Xbox One (Model 1708)

### OS Version
macOS Mojave 10.14.4 (18E226)

### Driver Version
0.16.11
I have had this issue for a while ... 

### Connection Method
Wired (AmazonBasics: Micro USB to USB-C Cable)

### Device Name and Info

**USB 3.0 Bus:**

  Host Controller Driver:	AppleUSBXHCISPT
  PCI Device ID:	0xa12f 
  PCI Revision ID:	0x0031 
  PCI Vendor ID:	0x8086 

**iBridge:**

  Product ID:	0x8600
  Vendor ID:	0x05ac (Apple Inc.)
  Version:	1.01
  Manufacturer:	Apple Inc.
  Location ID:	0x14200000

**Controller:**

  Product ID:	0x02ea
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	3.01
  Serial Number:	3032363030333738393633363434
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Microsoft
  Location ID:	0x14100000 / 10
  Current Available (mA):	500
  Current Required (mA):	500
  Extra Operating Current (mA):	0

### Response of Controller
Controller generally works fine and guide button is lighted. Though it is when playing a game on occasion the buttons will stop working and do nothing even after pressing them repeated x15 times; hence I die in game bc of it. I would say it happens frequently, but not everyday.

### Response in Preference Pane
Does the controller appear in the preference pane? YES
Do all of the buttons, joysticks, and triggers act as expected? YES

### Detailed Explaination of Issue
The most recent timestamp of this issue was at 8:55:44 PM EST. I was playing Super Animal Royale, and pressed A to use an item "duck tape", it did nothing, then I press it again and again nothing happened. I then had to move because I didn't want to die in game. I then had to reload so I pressed B, again nothing happened, I pressed it again and again. eventually it worked and my weapon reloaded. (Additionally, This happens in other games to Don't Starve Together, the A button a lot of the time won't work.)

I have no idea where the logs are .... or I would send them.

FranticRain, 2019-04-04:
Is it just the buttons that do not work or is it all of the inputs from the device? I.E. do the joysticks and triggers also fail to recognize input when the buttons stop working?

L0VEFIRE, 2019-04-04:
From my experience its only the buttons (A,B,X,Y)

FranticRain, 2019-04-06:
Okay, so here's why this is really weird: All of the output data from the controller returns to the driver in one big packet of data. Joysticks, triggers, all of the buttons, etc. This means that if the driver is dropping inputs on these buttons, it should be dropping inputs on EVERY input. At least from the driver's perspective. There are a couple of potential causes:

1. Your controller is somehow going bad and doesn't like those buttons anymore.
2. Your controller is occasionally sending some bad packets.
3. Your controller isn't sending a packet for a button press.

In order to sort of debug the issue, the next time you encounter this issue I need you to spin one of the sticks around wildly. Every time it moves, it generates a packet. So if you press a button while moving the stick, it will almost certainly get picked up in the flurry of packets caused by moving the joystick. I know this isn't an optimal solution, but it should help me get a feeling for what is going on.

L0VEFIRE, 2019-04-06:
Interesting... Okay will do. I will work on thiz and get back to you. Thanks.

FranticRain, 2019-10-24:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-10-24 by FranticRain