360Controller/360Controller #858: Controller does not show up in Preferences and allow developer message doesn't disappear

martymartymarty, 2019-04-05:
### Type of Controller
Replace this text with: Xbox One (Wired)

### OS Version
10.13.6 High Sierra

### Driver Version
0.16.11

### Connection Method
Wired

### Device Name and Info
Device Name: Controller
Product ID: 0x02ea
Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
Doesn't light up. Vibrates once when connected. Works fine on PC. 

### Response in Preference Pane
Controller does not appear in the preference pane.

### Detailed Explaination of Issue
During installation of 0.16.11 I pressed 'Allow' in Security & Privacy for the message 'System software from "Drew Mills" was blocked from loading'. After pressing Allow both the message and the button did not disappear, I'm not sure whether that's normal. 

The installation did complete and asked for a reboot. After the reboot I could not find the Controller icon in System Preferences, but it does show up in the System Report as a USB Controller. 

Security & Privacy still shows the message and the Allow button but clicking it doesn't make it disappear (should it?) nor does it cause the controller to show up in System Preferences.

Thanks in advance for your time!

FranticRain, 2019-04-06:
It looks like you're using some piece of software or hardware that macOS doesn't particularly like. Long story short: yes, that prompt should disappear. The reason that it isn't disappearing is that macOS thinks that some outside actor is trying to accept in on your behalf, and that the action is being taken by you, the user. So it is ignoring the click. This can be caused by software like Karabiner or hardware like a Cintiq tablet. If you have any sort of software or hardware that modifies your input, either unplug it or disable it and try to press the accept button.

Unfortunately, macOS isn't going to let the driver work until you can properly accept the certificate.

martymartymarty, 2019-04-06:
Thanks so much for the quick response @FranticRain. I disabled Wacom software and reinstalled but that didn't help the issue. However, via some Googling I ended up on the Karabiner Github where I found the solution. 

I had to press Allow _before_ doing the restart after completing installation. The message disappeared after pressing Allow and after the restart the software showed up in the Preference Pane. Hope this can help someone in the future. Thanks for your time and the driver!

FranticRain, 2019-04-07:
Ah. That's a good tip. I'll try and remember that one so I can pass it on to people with the same issue in the future.

closed 2019-04-07 by FranticRain