360Controller/360Controller #859: My Device not supported

bartoszswiety, 2019-04-06:
### Type of Controller
ThirdParty USB

### OS Version
Mojave + Xcode 10 

### Connection Method
USB WIRED

### Device Name and Info
>Device Name:Microntek       
<br/>Product ID:0x0006
<br/>Vendor ID:0x0079

### Detailed Explaination of Issue
Unfortunately kext is not building with my Xcode 10 (it's not supported right?) - can I ask for  adding my device and sending me a driver? 

Cheers!

FranticRain, 2019-04-06:
Quick question before I build this. Does this controller work with an Xbox console at all? Could you link me to a store listing for it? Like Amazon, or even the official page for it.

bartoszswiety, 2019-04-08:
https://www.amazon.fr/gp/product/B00546SWJQ

Not sure if this controller works with Xbox - I just want to fake Steam to treat this controller as xbox's. Here you can find my key's config:

https://github.com/libretro/retroarch-joypad-autoconfig/blob/master/udev/Microntek%20%20%20%20%20%20%20%20%20%20%20%20%20%20USB%20Joystick%20%20%20%20%20%20%20%20%20%20.cfg

FranticRain, 2019-04-14:
I don't expect this to work, but let me know how it works out: 
[Install360Controller.pkg.zip](https://github.com/360Controller/360Controller/files/3077768/Install360Controller.pkg.zip)

FranticRain, 2019-10-24:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-10-24 by FranticRain