360Controller/360Controller #860: Add New Controller ID

dolfelt, 2019-04-08:
### Type of Controller
Replace this text with: PDP XBox One Ghost White

### OS Version
10.14

### Driver Version
16.11

### Connection Method
Wired

### Device Name and Info
Device Name: PDP Wired Controller for Xbox One - Stealth Series | Ghost White
<br/>Product ID: 0x02a5
<br/>Vendor ID: 0x0e6f

### Response of Controller
This controller is not recognized.

### Response in Preference Pane
The controller is not recognized.

### Detailed Explaination of Issue
I understand that the PDP controllers don't seem to be supported, but when examining the code, it looks like the black version of this same controller is supported, but not the white one. I'd love some more information about this. Thank you!

FranticRain, 2019-04-14:
Fair warning. Just because it's in the plist doesn't mean that it works. Here's a build with your controller added. It may or may not work.
[Install360Controller.pkg.zip](https://github.com/360Controller/360Controller/files/3077771/Install360Controller.pkg.zip)

closed 2019-04-14 by FranticRain

dolfelt, 2019-04-15:
Thank you so much @FranticRain, it works! I'm using macOS 10.14.3