360Controller/360Controller #861: Installer is not ins

Sadrat12, 2019-04-08:
### Type of Controller
Xbox 360
### OS Version
Replace this text with the version of macOS that you are running

### Driver Version
Replate this text with the version or versions of the driver that you are using or have tested

### Connection Method
Replace this text with: Wired, Wireless Adapter, Bluetooth, N/A or more specific method if applicable

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID:
<br/>Vendor ID:

### Response of Controller
Replace this text with information on how the physical controller reacts. Does the guide button light up? Does it vibrate? Etc.

### Response in Preference Pane
Replace this text with information about the controller's reaction in the preference pane. Does the controller appear in the preference pane? Do all of the buttons, joysticks, and triggers act as expected?

### Detailed Explaination of Issue
Replace this text with a detailed explaination of your issue. Simply saying "it doesn't work" is a quick way to ensure that your issue is never fixed. Without detailed information on your specific issue, we cannot fix your issue. Including as much information as possible in this section increases the likelyhood of getting your issue quickly and effectively resolved.

FranticRain, 2019-04-14:
Skipping the "Destination Select" step is normal, because kexts and preference panes can only be installed in one location.

The installation button should create a popup that prompts for administrator privileges. Are you missing it, or do you not have administrator rights on your machine?

FranticRain, 2019-10-24:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-10-24 by FranticRain