360Controller/360Controller #865: XBox Controller S is not appearing in the combolist under 10.14.4 (Hackintosh)

friguron, 2019-04-22:
### Type of Controller
Original XBox Controller S (Wired).

### OS Version
Mojave 10.14.4 (It's a properly working Hackintosh otherwise)

### Driver Version
10.16.11 (but in fact it never worked in any of my hackintoshes in the past for the last 2 years... I just never decided to report it).

### Connection Method
Wired, with typical adapter.

### Device Name and Info
Device Name: Hub:
Product ID: 0x0288
Vendor ID: 0x045e
I'm unsure if I targeted the right USB object but it was the only one from Microsoft in the Apple Report device list.

### Response of Controller
It does nothing, 0. It doesn't even gets detected. No vibration.

### Response in Preference Pane
The combo to select it appears as empty/no device detected. No further action can be performed.

### Detailed Explaination of Issue
I just installed your 10.16.11 driver version into a fresh Mojave (otherwise perfectly working) Vanilla Hackintosh instalation. Everything went ok apparently, no complain about the installation process, no complain about having to give permissions for the driver to be loaded... The reset was performed right after the installation.

If I type this:  
kextstat | grep 360
I get this:
165    0 0xffffff7f84696000 0xc000     0xc000     com.mice.driver.Xbox360Controller (0.16.11) D4B7CD21-E160-348D-85FA-F31987063A9A <80 50 6 5 3>

If I type this:
sudo kextload 360Controller.kext
I get nothing at all.

I assume the driver is loaded then (?)

I've tried to connect my XBOX controller S adapter to any USB port in my hackintosh (USB 2 and 3), and nothing happens. Reboots and resets, same thing. All these USB ports OTOH work perfectly with any other USB device I connect to.

I'm out of ideas. I have to say that this Controller has always worked perfectly under Windows XP with Tiger "Random" driver. In fact I just checked it 5 minutes before.

I will report and test your suggestions as soon as I get them. Greetings.

FranticRain, 2019-04-22:
Tagging @kasbert, as he implemented the original Xbox controller support for this driver.

friguron, 2019-04-22:
I've just browsed some other open threads, and apparently my situation is exactly the same as the one describe in this other thread:

https://github.com/360Controller/360Controller/issues/851

That is, if I boot my Mojave Hackintosh with the pad inserted into the USB port, the driver detects it perfectly. But as soon as you remove the controller from the USB port, (and then reinsert it again), the driver won't recognize it anymore in that MacOS session. You need a reset WITH THE PAD INSERTED for the driver to recognize it (and then don't disconnect it).

You can merge the threads if needed. No need for two open threads.

closed 2019-04-23 by FranticRain