360Controller/360Controller #866: 360 Wired Controller - No Devices Found

MarkRyanDev, 2019-04-27:
### Type of Controller
 Xbox 360

### OS Version
Mojave 10.14.4 (18E226)

### Driver Version
Version 0.16.11

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Controller
<br/>Product ID: 0x028e
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
When plugged in, the whole circle flashes green continuously

### Response in Preference Pane
No Devices Found.

### Detailed Explaination of Issue
I had this exact same setup on my old Macbook Pro and it worked fine. I am using a USB-C adaptor. I clicked "allow" in security settings. I have reinstalled several times. I ran the `kextutil` command and got this 
```
$ sudo kextutil /Library/Extensions/360Controller.kext
Password:
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings:
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done):
        ChatPadKeyboardEvents
```

FranticRain, 2019-04-27:
There have been some issues with USB-C adapters previously. We haven't quite found a solution to the issue, but could you give [1.0.0-alpha.1](https://github.com/360Controller/360Controller/releases/tag/v1.0.0-alpha.1) a shot?

MarkRyanDev, 2019-04-27:
Ok, so I turns out I was actually already using the 1.0 alpha (it still says Version 0.16.11 in the About page). I tried downgrading to actual 0.16.11 and it works!

closed 2019-04-27 by MarkRyanDev

FranticRain, 2019-04-27:
That means that wired Xbox 360 controllers are probably messed up in 1.0.0-alpha.1. That's good to know.