360Controller/360Controller #868: 1.0a1: Disabling driver leads to hard crash

PokeCaptain, 2019-04-27:
### Type of Controller
Xbox One wired

### OS Version
10.14.5b3

### Driver Version
1.0a1

### Connection Method
Wired
### Device Name and Info
n/a

### Response of Controller
N/A

### Response in Preference Pane
Hard Crash

### Detailed Explaination of Issue
Had my controller plugged in and was testing the preference pane of the new version. Attempting to disable the driver as a test led to System Preferences hard crashing with the following report:

`Process:               System Preferences [11087]
Path:                  /Applications/System Preferences.app/Contents/MacOS/System Preferences
Identifier:            com.apple.systempreferences
Version:               14.0 (14.0)
Build Info:            SystemPrefsApp-412000000000000~792
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           System Preferences [11087]
User ID:               501

PlugIn Path:             /Library/PreferencePanes/Pref360Control.prefPane/Contents/MacOS/Pref360Control
PlugIn Identifier:       com.mice.driver.360Controller.Prefs
PlugIn Version:          0.16.11 (0.16.11)

Date/Time:             2019-04-27 15:08:56.618 -0400
OS Version:            Mac OS X 10.14.5 (18F118d)
Report Version:        12
Anonymous UUID:        C2411396-0A84-ED0C-FEE4-EC4D47B8FF86

Sleep/Wake UUID:       6D95A465-3358-48DE-A00D-946E51AFA885

Time Awake Since Boot: 28000 seconds
Time Since Wake:       5800 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [11087]

Application Specific Information:
com.mice.driver.360Controller.Prefs v.0.16.11 (Xbox 360 Controllers)
Performing @selector(toggleDriverEnabled:) from sender NSButton 0x6000006fff40

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.apple.ForceFeedback       	0x00007fff430c68b2 FFDeviceEscape + 29
1   com.mice.driver.360Controller.Prefs	0x00000001133e9da5 -[Pref360ControlPref powerOff:] + 65
2   com.mice.driver.360Controller.Prefs	0x00000001133e9ada -[Pref360ControlPref toggleDriverEnabled:] + 145
3   com.apple.AppKit              	0x00007fff3e7f8920 -[NSApplication(NSResponder) sendAction:to:from:] + 312
4   com.apple.AppKit              	0x00007fff3e862c6e -[NSControl sendAction:to:] + 86
5   com.apple.AppKit              	0x00007fff3e862ba0 __26-[NSCell _sendActionFrom:]_block_invoke + 136
[... 373 lines cut ...]
===========                     =======  ======= 
TOTAL                              1.1G      892 
TOTAL, minus reserved VM space   790.3M      892 

Model: MacBookPro12,1, BootROM 184.0.0.0.0, 2 processors, Intel Core i5, 2.7 GHz, 8 GB, SMC 2.28f7
Graphics: kHW_IntelIris6100Item, Intel Iris Graphics 6100, spdisplays_builtin
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1867 MHz, 0x80AD, 0x483943434E4E4E424C54414C41522D4E5544
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1867 MHz, 0x80AD, 0x483943434E4E4E424C54414C41522D4E5544
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x133), Broadcom BCM43xx 1.0 (7.77.61.2 AirPortDriverBrcmNIC-1305.8)
Bluetooth: Version 6.0.12d18, 3 services, 27 devices, 1 incoming serial ports
Network Service: Thunderbolt Ethernet, Ethernet, en6
Network Service: Wi-Fi, AirPort, en0
PCI Card: Apple 57762-A0, Ethernet Controller, Thunderbolt@195,0,0
Serial ATA Device: APPLE SSD SM0256G, 251 GB
USB Device: USB 3.0 Bus
USB Device: Bluetooth USB Host Controller
USB Device: USB2.0 Hub
USB Device: Controller
USB Device: USB Optical Mouse
USB Device: USB2.0 Hub
USB Device: EB1
USB Device: Canon CAPT USB Device
Thunderbolt Bus: MacBook Pro, Apple Inc., 27.1
Thunderbolt Device: Thunderbolt to Gigabit Ethernet Adapter, Apple Inc., 1, 5.5
`

FranticRain, 2019-04-27:
Hmm. I can't reproduce this on 10.14.4. I'll keep looking into it.

FranticRain, 2019-04-28:
Just adding some notes so I don't forget:

According to the kernel logs, the failure occurred in `Pref360ControlPref::powerOff`. Specifically, on the call to `FFDeviceEscape`. The termination code implies that something was null at the time of the call, but `ffDevice` was checked for 0 value before this line, so if it was properly de-alloced, it should have been nulled out and this would be avoided. `escape` clearly isn't null because it's a function scope object. So either `ffDevice` got de-alloced out from under us, or we are de-allocing it somewhere and not setting it to null. As far as I can tell, we aren't de-allocing `ffDevice`, so it must have been de-alloced out from under us? Why?

FranticRain, 2019-05-14:
I've managed to recreate a preference pane crash in the stable 10.14.5 release. Can you confirm whether or not you can still recreate this bug in 10.14.5 stable?

PokeCaptain, 2019-05-15:
Just got 10.14.5 stable. Will test in the morning

PokeCaptain, 2019-05-17:
I can't actually reproduce this anymore, which kinda sucks. I found another bug in the meantime though, which will be a separate issue.

FranticRain, 2019-05-17:
If you can't repro, I have to assume that it was a kernel issue and has been fixed.

closed 2019-05-17 by FranticRain