360Controller/360Controller #869: Xbox 360 wired controller does not show up using alpha driver

uvesten, 2019-04-29:
### Type of Controller
Xbox 360

### OS Version
10.14.4 (18E226)
Hardware Overview:
Model Name:	MacBook Pro
 Model Identifier:	MacBookPro15,2

### Driver Version
v1.0.0-alpha.1

### Connection Method
Wired

### Device Name and Info

>Device Name: Controller
<br/>Product ID: 0x028e
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
The green ring around the center button blinks a few times, then turns black. 

### Response in Preference Pane
The controller does not show up in the preference pane. 

### Detailed Explaination of Issue
The controller does not show up either in the preference pane or in Dolphin emulator.

uvesten, 2019-04-29:
After uninstalling the alpha driver, and installing version 360ControllerInstall_0.16.11_Notarized.dmg instead, the controller works as expected.

FranticRain, 2019-04-29:
I'm aware of this issue, but I'm not quite sure why it's happening, since Xbox One controllers work properly. I'm still looking into it, but I'm going to keep this one open to mark it.

PokeCaptain, 2019-04-29:
See #866

FranticRain, 2019-04-29:
With the alpha version installed, could you open up `Console.app` and put `process:kernel` (don't forget to press enter) into the search bar? This will give you just the kernel logs, which will include the logs from the driver. When you connect the controller, it may spit out some mangled logs. If you see anything that might be pertinent, even if it is mangled, post it here. That will help me figure out what is going wrong.

uvesten, 2019-04-29:
Here you go!

```
default	18:42:42.366331 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366343 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366347 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366352 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366357 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366362 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366367 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366371 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366376 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366394 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366423 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366441 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366470 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366492 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366511 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366530 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366549 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366567 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366586 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366604 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366628 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366656 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366675 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366694 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366714 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366733 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366752 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366766 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366776 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366799 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366827 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366846 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366860 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366879 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366898 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366916 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366934 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366953 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366975 +0200	kernel	Controller: family specific matching fails
default	18:42:42.366994 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367013 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367032 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367041 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367053 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367075 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367081 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367088 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367095 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367102 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367108 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367115 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367122 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367129 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367135 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367139 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367145 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367149 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367154 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367159 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367165 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367172 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367179 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367186 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367192 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367199 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367206 +0200	kernel	Controller: family specific matching fails
default	18:42:42.367210 +0200	kernel	Controller: family specific matching fails
[... 20 lines cut ...]
default	18:42:42.367304 +0200	kernel	Controller: family specific matching fails
[... cut ...]
default	18:42:42.367400 +0200	kernel	Xbox360Peripheral::probe(Controller)
default	18:42:42.367406 +0200	kernel	Xbox360Peripheral::start(Controller) <1>
default	18:42:42.367410 +0200	kernel	Hello world.
default	18:42:43.075057 +0200	kernel	Pipe type: 1, pipe direction: 3
default	18:42:43.075427 +0200	kernel	Pipe type: 0, pipe direction: 3
default	18:42:43.075836 +0200	kernel	start - Unable to find chatpad in pipe.
default	18:42:43.075848 +0200	kernel	Fail.
default	18:42:43.075943 +0200	kernel	Xbox360Peripheral::start(Controller) <1> failed
```

FranticRain, 2019-04-29:
Perfect. I'll have a build up later today that *should* solve this problem. Or at least will help me get headed in the right direction. Not sure why it's failing when the logic flow is theoretically the same as the old code though. Strange.

FranticRain, 2019-04-30:
Okay, does this fix the issue? If it does, let me know. If it doesn't, post the logs again. Feel free to leave out all of the "family specific matching fails" noise. That's just macOS being a punk.

[360ControllerInstall_1.0.0-alpha.2.dmg.zip](https://github.com/360Controller/360Controller/files/3129684/360ControllerInstall_1.0.0-alpha.2.dmg.zip)

JayRabxx, 2019-04-30:
FWIW, I was having a similar issue and the alpha 2 file you linked fixed it!

uvesten, 2019-04-30:
This release does indeed fix the issue! Controller is working as it should. 

Just FYI, here are the logs using alpha 2:

```
default	11:29:33.041619 +0200	kernel	Xbox360Peripheral::probe(Controller)
default	11:29:33.041623 +0200	kernel	Xbox360Peripheral::start(Controller) <1>
default	11:29:33.041626 +0200	kernel	Hello world.
default	11:29:33.044524 +0200	kernel	Pipe type: 1, pipe direction: 3
default	11:29:33.044805 +0200	kernel	Pipe type: 0, pipe direction: 3
default	11:29:33.045103 +0200	kernel	Chatpad pipe type: 1, pipe direction: 3
default	11:29:33.045106 +0200	kernel	start - Unable to find chatpad in pipe.
default	11:29:33.045138 +0200	kernel	Pass.
```

FranticRain, 2019-04-30:
Perfect. I'll close this issue out and tag it when I get a chance.

FranticRain, 2019-05-01:
Tagged into a pre-release. Closing.

closed 2019-05-01 by FranticRain