360Controller/360Controller #870: I need help with my controller

quinten9966, 2019-04-30:
### Type of Controller
Wireless xbox 360 

### OS Version
Version 10.13.6

### Driver Version
lastest version and 0.16.5

### Connection Method
bluetooth

### Response of Controller
nothing only turning on and when I press the button to connect it changes 

### Response in Preference Pane
it doesn't connect

### Detailed Explaination of Issue
i want to connect my controller but I doesn't work I don't have anything else then the controller I don't have an cable or anything else please let me know what I need to use to make it work and how I need to let it work

FranticRain, 2019-04-30:
Xbox 360 controllers don't use Bluetooth, and they aren't currently supported by this driver.

closed 2019-04-30 by FranticRain

quinten9966, 2019-04-30:
what driver do I need for the controller ?

FranticRain, 2019-04-30:
Some users have said that version 0.14 of this driver works with the wireless controller, as long as you have a wireless dongle. I have also heard that it causes kernel panics. Use old versions of the driver at your own risk.

quinten9966, 2019-04-30:
do i also need to use receiver ?

quinten9966, 2019-04-30:
and what are kernel panics ?

FranticRain, 2019-04-30:
You have to use a receiver in order to use an Xbox 360 controller wirelessly. Kernel panics are hard crashes of the operating system. Like the "blue screen of death" in Windows.

quinten9966, 2019-04-30:
ohh okay thanks for telling me and I am going to order the receiver from amazon if I have the receiver what do I got to do then ?

FranticRain, 2019-04-30:
You would have to install a version of this driver that supports wireless controllers. I believe that 0.16.5 is the most recent version that supports wireless controllers.

quinten9966, 2019-04-30:
and then connect with the program and then I good to go right ?

FranticRain, 2019-04-30:
Theoretically, yes. There are no guarantees.

quinten9966, 2019-04-30:
ohh okay well thx anyways

quinten9966, 2019-05-09:
Hey I got the receiver what do I do now ? I also have a disk

FranticRain, 2019-05-09:
Install 0.16.5

quinten9966, 2019-05-09:
okay I did that and my receiver's light is green

quinten9966, 2019-05-09:
and do I have to do anything with the disk ?

FranticRain, 2019-05-09:
There is no need for the disk. You are done. If it doesn't work now then you're outside the realm of support, as I only support the latest version and wireless controllers aren't supported in the latest version.

quinten9966, 2019-05-09:
so what do I do now then ? cause I have the receiver and the wireless controller

FranticRain, 2019-05-09:
Answers to basic questions like these are in the README, or can be answered by the standard Xbox 360 controller documentation: https://support.xbox.com/en-US/xbox-on-windows/accessories/xbox-controller-for-windows-setup#2727833d5f964641a2236ffa366c0bd3

quinten9966, 2019-05-09:
well the README only says about errors and the link doesn't say anything about Mac but when I plug the receiver in and turn on my controller is doesn't connect to the receiver do you know the problem for the or do you know where I can find that ?

quinten9966, 2019-05-09:
is there any other controller that is easier to setup for Mac ?