360Controller/360Controller #871: MYGT C03 - detected but can't pair with a dongle

foxnoodles, 2019-05-01:
### Type of Controller
Third party 

### OS Version
MacOS 10.14.4

### Driver Version
0.16.11

### Connection Method
Wireless Adapter

### Device Name and Info
MYGT C03, which has a 2.4 MHz dongle and 4 input modes (XInput, PS3, Android, IOS) that can be changed on the fly. 
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: identifies as "Controller(XBOX 360 For Windows)"
<br/>Product ID: 0x028e
<br/>Vendor ID: 0x045e

### Response of Controller
None

### Response in Preference Pane
Pref pane detects the dongle as "XBox 360 Wired Controller (wired)" that's all

### Detailed Explaination of Issue
Ok so, the same dongle works fine on windows, the controller gets recognized as an official XBOX 360 controller and works like charm. But on Mac, the driver only detects the dongle but the controller itself can pair to it. (this is indicated by blinking green light). 
It is worth to note that system report calls it  "Controller(XBOX 360 For Windows" no matter what. Even if the 360 driver is disabled. 

Any ideas?

foxnoodles, 2019-05-01:
Okay I'm an idiot

"**Xbox One Controllers connected with Wireless Adapter**
Xbox One controllers connected with the Wireless Adapter are currently not supported. Please be patient as we figure out this complicated protocol."

closed 2019-05-01 by foxnoodles

FranticRain, 2019-05-01:
You should be able to use this controller with macOS with Bluetooth at least. It won't work with the driver, and game support might be poor, but at least it will work.

foxnoodles, 2019-05-01:
Yeah but it detects it as a keyboard not just a "bluetooth device"

FranticRain, 2019-05-01:
That's kinda crappy. It might work as a wired controller because it steals the VID and PID of the Xbox One controller. But if it doesn't, it's outside our purview of support as we do not support "counterfeit" controllers such as this one that violate the USB specification by stealing another device's VID/PID.

foxnoodles, 2019-05-01:
Yeah I figured. 
Also, I think the dongle fakes itself to be the wired Xbox controller, while the gamepad itself has a different PID. At least that's how it shows up when paired via Bluetooth.  I just don' get why it doesn't pair with an adapter.

Here's the Bluetooth device that acts like a keyboard:
 
`MYGT Controller:
  Address:	FF-0F-80-02-63-25
  Major Type:	Peripheral
  Minor Type:	Keyboard
  Services:	Bluetooth Gamepad   
  Paired:	Yes
  Configured:	Yes
  Connected:	No
  Firmware Version:	0x011B
  Vendor ID:	0x4D59
  Product ID:	0x4303
  Class of Device:	0x05 0x10 0x2540
  Host Connectable:	No
  EDR Supported:	No
  eSCO Supported:	No
  SSP Supported:	Yes`

foxnoodles, 2019-05-01:
Okay, this is weird, I hooked up the dongle to VM on windows to let it pair. Then paused the VM. It stayed in paired state and now works on Mac

alypasya, 2021-07-19:
Hai bro, i have problem with mygt c03, when i connect to windows the lamp blinking green and yellow, which is windows known as ps3 controller and did not recognized as x input. Any solution?

melon-husk, 2022-12-16:
I think I have the same MYGT controller, it only works if I set it to android mode on the controller and connect through bluetooth on pop os.