360Controller/360Controller #873: Alpha 2 not running on 'Intel based mac'. Previous versions not detecting controller

davidd20, 2019-05-02:
### Type of Controller
Xbox 360 Wired Controller

### OS Version
OSX 10.13.6

### Driver Version
1.0.0-alpha.2. Have also tried 1.0.0-alpha.1, and 0.16.11

### Connection Method
Wired

### Device Name and Info
Device Name: Controller
Product ID: 0x028e
Vendor ID: 0x045e (Microsoft Corporation)

### Response of Controller
None

### Response in Preference Pane
The controller appears in the system report of the mac but it isn't detected in the pref pane.

### Detailed Explaination of Issue
My main issue with the Alpha 2 release is that the pref pane won't even open. It says it can't be run on an intel-based mac. My issue with the older versions is that they don't detect my controller. I've tried all usb ports and can't think of anything else.

davidd20, 2019-05-02:
Still haven't figured out the issue with alpha.2 but I installed version 0.16.9 and it detects my controller. Maybe later versions require a later OSX?

FranticRain, 2019-05-02:
The alpha versions are built against the 10.14 SDK, so it wouldn't completely surprise me if it doesn't work on 10.13.X for some reason. But the fact that alpha1 does and alpha2 doesn't is a bit concerning to me. I'll build against the 10.11 SDKs and upload it when I get a chance and you can let me know if it fixes your issue.

FranticRain, 2019-05-03:
Didn't bother to notarize this one since you're running 10.13 anyway. Let me know how this works out for you.

[Install360Controller.pkg.zip](https://github.com/360Controller/360Controller/files/3140366/Install360Controller.pkg.zip)

davidd20, 2019-05-03:
Thanks a lot for getting on this so fast! That one works 😁

closed 2019-05-03 by davidd20

FranticRain, 2019-05-03:
Great. I'll put out an updated version of alpha2 that is all notarized up when I get a chance.

FranticRain, 2019-05-05:
I wasn't super satisfied with the way I fixed this before, so could you give this installer a shot real quick? It should install a different version depending on what version of macOS you have installed. So people on 10.14 can get 10.14 specific features, etc.

[Install360Controller.pkg.zip](https://github.com/360Controller/360Controller/files/3145033/Install360Controller.pkg.zip)