360Controller/360Controller #874: Remapping?

solomani, 2019-05-04:
This isn't an issue but a question and I couldn't find an answer on the wiki.  How does remapping work?

I hit **Start Remapping**, the up stick gets highlighted but none of my inputs are accepted - it just beeps.  

Thanks.

FranticRain, 2019-05-04:
The way remapping should work is that the application will highlight a button on the controller, and then you press the button that you would like to emit that value. So if you want to, say, invert the D-Pad for some reason, when D-Pad up is highlighted, you would press down, etc.

Make sure that your controller is selected and that you are getting proper reading of the inputs on the main screen. Also note that you can only remap controller buttons to other controller buttons. You cannot map buttons to say, inputs on your keyboard.

Let me know if you need any more information.

solomani, 2019-05-05:
Thanks FranticRain.  I was hoping to use it to remap keyboards keys so I could play something like diablo2 with a controller, but sounds like I can't do this?

FranticRain, 2019-05-05:
That is correct. Remapping controller inputs to keyboard inputs is much more effective and much easier in a user space application. Therefore, we avoid that functionality in the driver. These days I usually recommend tools like [Enjoyable](https://yukkurigames.com/enjoyable/) for this purpose.

solomani, 2019-05-05:
Perfect, thank you.

closed 2019-05-05 by FranticRain