360Controller/360Controller #875: Help please! Xbox One controller won't work on driver or bluetooth for the latest Mac 2018 Mojave??

csagastume1, 2019-05-04:
Type of Controller
Xbox One- red controller

### OS Version
Mojave macOS
10.14.4
### Driver Version
10.6,5, alpha 1.0...

### Connection Method
Wired and Bluetooth

### Device Name and Info

<br/>Product ID:0x02ea

<br/>Vendor ID:0x045e  

### Response of Controller & Preference Pane
Wired Version: When I plug in the wire, the remote vibrates once and no lights or remote come on. Ive tried downloading drivers, but when I get to the "no remote found", no remote shows up. pressing buttons on remote does not make the on screen remote move. 

Bluetooth: hooks up great. remote lights on. but when I start fortnite up, the buttons start to flicker on my screen and the only button that functions is "B". and it seems that it acts like "A" because it will continue to purchasing packages if I keep pressing it.

FranticRain, 2019-05-04:
Re: Bluetooth:
Bluetooth controllers are not supported by this driver. They will not show up in the preference pane of this driver. Bluetooth Xbox One controllers are supported by default in the operating system, and therefore do not need a driver to function. You only need the driver if you wish to connect your Xbox One controller via USB. If you would like to test if your Xbox One Bluetooth controller is properly connected to your computer, I recommend a HTML5 gamepad tester like [this one.](http://html5gamepad.com/)

Re: Wired
First, I want to make sure that the driver is working correctly on your machine, so open up the terminal and run this command, entering your password as prompted:
```
sudo kextutil /Library/Extensions/360Controller.kext`
```
It shouldn't matter which version you have installed, but I would stick with 0.16.5 for now since it is the stable release. Let me know what the output of that command is.

csagastume1, 2019-05-04:
Awesome got it to work! Thanks! I just had to go to allow in settings for drew mills

closed 2019-05-05 by FranticRain