360Controller/360Controller #876: "Xbox 360 Controllers" preference pane crashes

nicocrouzet, 2019-05-05:
### Type of Controller
N/A

### OS Version
10.12.6 (16G1918)

### Driver Version
1.0.0-alpha.2

### Connection Method
N/A

### Device Name and Info
N/A

### Response of Controller
N/A

### Response in Preference Pane
Preference pane crashes when I click on "Xbox 360 Controllers".

### Detailed Explaination of Issue
Nothing special to add expect that it crashes. I don't know if it can helps, but here is the console output when crashing :
[console.pdf](https://github.com/360Controller/360Controller/files/3145908/console.pdf)
Please let me know if you need more information (and thanks for providing this driver to everyone for free!!!).

FranticRain, 2019-05-05:
I'm aware of this issue and I've been working on a fix, can you give this installer a shot and let me know if it fixes the issue? 
[Install360Controller.pkg.zip](https://github.com/360Controller/360Controller/files/3146151/Install360Controller.pkg.zip)

Kiloutre, 2019-05-06:
I had the same issue as him, also under 10.12.6m and the installed you linked fixes the issue.

nicocrouzet, 2019-05-06:
@FranticRain 
Thanks for your quick feedback and for the fix!
Just tried and confirm it resolved the issue!
Now I can open the preference pane and my xbox one controller is recognized!
When will this fix be released officially?
Thanks.

FranticRain, 2019-05-06:
Good to hear that this is fixed. I'll go ahead and keep this open as a source for a fix for this issue.

@nicocrouzet 
This release is just as official as any of the "alpha" releases, but I hope to have an alpha-3 out in a week or two. It's tied up in another feature that was halfway done when I implemented this fix.
If you're talking about 1.0.0 in full release, that will probably be another two months or so. There is still a lot to be done, and my progress is fairly slow with a full time job to contend with now.

nicocrouzet, 2019-05-06:
@FranticRain ,
I was talking about an alpha 3. But as the version you sent to me works, it is okay for me and I will update once a new release is available.
I imagine that there is still a lot of work before 1.0.0 is released!

**Anyway, thanks again for your great and reactive support, and for your contribution of making our XBox controllers working on macOS !!!**

closed 2019-10-24 by FranticRain